knowrob/knowrob #66: assertion of facts through json-prolog relatively slow

airballking, 2015-10-19:
Abstract: 
It seems assertion of facts into ```knowrob``` through the ```json-prolog``` interface takes about 0.2s per fact. I need to speed this up. @daniel86 @knowrob @fairlight1337 Could you please help me make with this?

Details:
I am trying to assert new facts into a running ```knowrob``` instance runtime to later query for those facts through the ```json-prolog``` interface. I have the impression that these assertions are relatively slow. 

For example, I am logging a perception instance with query and result designators from ```CRAM``` through the ```knowrob-plugin``` of ```semrec```. The actual call to the perception, and asynchronous logging calls to ```semrec``` in the ```repl``` feel like they are instantaneous. However, the info printouts of the ```knowrob-plugin``` asserting into the live knowrob take several seconds. I can literally watch the facts going into the knowledge base.

I want to use the ```knowrob``` logging for maintaining my belief state outside of ```CRAM```. Hence, I'm afraid several seconds to remember one perception event might be too slow for my requirements.

In order to verify that it is not the ```knowrob-plugin``` of ```semrec``` that takes so long, I ran a naive benchmark from the ```repl``` asserting the same fact 100 times through ```json-prolog``` and measuring the execution time. It turns out, one assertion tasks almost always 0.2s. This fits to my feeling of several seconds for logging one perception events because that should be around one or two dozen assertions. 

Do you guys have any idea how to speed up this process?

Appendix:
For future reference, I am putting the lisp code I used for my naive benchmarking below:

Naive benchmarking:
```lisp
(let ((before nil)
       (after nil))
  (loop for i from 1 to 100 do
      (setf before (cons (ros-time) before))
      ;; here comes the actual call to json-prolog
      (prolog-simple
        "cram_start_action(knowrob:'WithFailureHandling', 'WITH-FAILURE-HANDLING', 1445243660.613, '_', ACTIONINSTANCE)")
      (setf after (cons (ros-time) after)))
  (defparameter *benchmarks* (mapcar #'- after before)))
```

Analysis of naive benchmarking:
```lisp
(let ((min (reduce #'min *benchmarks*))
       (max (reduce #'max *benchmarks*))
       (avg (/ (reduce #'+ *benchmarks*) (length *benchmarks*))))
  (format t "~%min: ~a, max: ~a, avg: ~a~%" min max avg))

min: 0.20276284217834473d0, max: 0.2043600082397461d0, avg: 0.20356736898422242d0
```

daniel86, 2015-10-19:
It could be an issue with multithreaded rosjava.
Each query is processed in a dedicated thread (queries must be bound to a thread).
For synchronization there are some Thread.sleep(100) and `synchronized` calls, maybe that's the issue.
You can find source with some debugging symbols and I changed Thread.sleep(100) to Thread.sleep(50) in [1,2]. Please try and report back.
  [1] -> json_prolog/src/main/java/org/knowrob/json_prolog/query/ThreadedQuery.java
  [2] -> json_prolog/src/main/java/org/knowrob/json_prolog/JSONPrologNode.java
[1] http://pastebin.com/uMxcn3DS
[2] http://pastebin.com/y1FENG8a

airballking, 2015-10-19:
I ran the same test with your adjusted calls to ```Thread.sleep(50)```. The results I get on the lisp side are much faster:

```lisp
min: 0.10398387908935547d0, max: 0.11527895927429199d0, avg: 0.10653826713562012d0
```

I'd say the speed-up is almost 100ms. Any idea what a safe lower bounds for these sleeps can be?

airballking, 2015-10-19:
I just further reduced both sleep-calls to 10ms. Here are the results on the lisp side for asserting 100 facts:

```lisp
min: 0.022568941116333008d0, max: 0.10052704811096191d0, avg: 0.02706268310546875d0
```

It looks like we can get even faster for the minimum and average case. I keep on trying.

airballking, 2015-10-19:
When reducing the sleeps to 1ms we seem to see the real calculation times. With the recent speed up, I allowed myself to run 1000 fact assertion. ;)

For the analysis, I extended my calculations to also calculate min, max, and avg for the middle 2 quantiles of the measurements to discard outliers. Here are the analysis results:

```lisp
GLOBAL: min=0.002827882766723633d0, max=0.011059045791625977d0, avg=0.004324703693389892d0

MID-QUANTILES: min=0.003955841064453125d0, max=0.0046770572662353516d0, avg=0.0044373621940612794d0
```

So, I'd say that worked pretty well. On average, we need around 5ms for asserting a single fact. Occasionally, this can shoot up to 11ms. All acceptable for me. Other speed-ups could come from batching asserts in the ```knowrob-plugin```. @daniel86 @knowrob @fairlight1337 any objections to reducing those sleeps to 1ms?

APPENDIX:
```lisp
(defun analyse-benchmarks (data)
  (labels ((calc-stats (data)
             (list
              (reduce #'min data)
              (reduce #'max data)
              (/ (reduce #'+ data) (length data))))
           (print-stats (prefix min max avg)
             (format t "~%~a: min=~a, max=~a, avg=~a~%" prefix min max avg))
           (sort-copy (data)
             (sort (copy-list data) #'<))
           (mid-quantiles (data)
             (subseq (sort-copy data) (/ (length data) 4) (* 3 (/ (length data) 4)))))
    (apply #'print-stats "GLOBAL" (calc-stats data))
    (apply #'print-stats "MID-QUANTILES" (calc-stats (mid-quantiles data)))))
```

airballking, 2015-10-19:
I opened up a pull request with the reduced sleeps for convenient merging after we have all green lights. :)

fairlight1337, 2015-10-19:
No problem here.

daniel86, 2015-10-19:
Request is merged, thanks!
Sleep calls can also be completely avoided by using sleep/wake up mechanism, I will look into this when I find the time.

closed 2015-10-19 by daniel86