knowrob/knowrob #77: Issues with knowrob_srdl:compile when instrucctions for knowrob installation

alquimist, 2015-12-15:
I've been trying install the CRAM language, when I process catkin_make in ros to install knowrob from stacks, there is a failure with some depencencies,  the error is show below, some body could provide any guide to install it,  

Thanks. 

Gradling tasks for knowrob_srdl
Catkin generate tree: replacing older version of gencpp[0.5.2->0.5.3]

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':knowrob_srdl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_srdl:knowrob_srdl:0.1.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [stacks/knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl] Error 1
make[1]: *** [stacks/knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
warning: [options] bootstrap class path not set in conjunction with -source 1.6
1 warning

FAILURE: Build failed with an exception.

Surxz, 2016-01-22:
This issue seems to be connected to the rosjava installation. The way I resolved it was to install rosjava from apt-get instead of building it from source. Afterwards just do a clean build of knowrob and it should work fine.

knowrob, 2016-01-25:
Indeed, the rosjava source version is quite tricky, especially the message generation. I recommend to use the rosjava .deb version whenever possible. Using the source version, multiple compilation runs of packages along the dependency chain sometimes worked, but I haven't figured out a 'clean' strategy :(

srinimd2005, 2016-12-20:
Hi 
I am facing the same problem from building Knowrob. Even I installed the rosjava through the .deb. 
It is still giving me the same error

* What went wrong:
Could not resolve all dependencies for configuration ':semantic_map_to_owl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.semantic_map_to_owl:semantic_map_to_owl:1.0.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob_addons/semantic_map_to_owl/CMakeFiles/gradle-semantic_map_to_owl] Error 1
make[1]: *** [knowrob_addons/semantic_map_to_owl/CMakeFiles/gradle-semantic_map_to_owl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':knowrob_cad_parser:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_cad_parser:knowrob_cad_parser:1.0.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob_addons/knowrob_cad_parser/CMakeFiles/gradle-knowrob_cad_parser] Error 1
make[1]: *** [knowrob_addons/knowrob_cad_parser/CMakeFiles/gradle-knowrob_cad_parser.dir/all] Error 2

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':json_prolog:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.json_prolog:json_prolog:0.1.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob/json_prolog/CMakeFiles/gradle-json_prolog] Error 1
make[1]: *** [knowrob/json_prolog/CMakeFiles/gradle-json_prolog.dir/all] Error 2

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':knowrob_srdl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_srdl:knowrob_srdl:0.1.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl] Error 1
make[1]: *** [knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl.dir/all] Error 2

Can someone suggest some help

asilx, 2016-12-20:
Hi,

Did you install swi-prolog and swi-prolog-java?

srinimd2005, 2016-12-20:
Yes I installed them.

daniel86, 2017-03-08:
@srinimd2005 Please re-open if you still have issues

closed 2017-03-08 by daniel86

munu8, 2017-03-15:
@srinimd2005 ,Have you solved the problem?I meet the same problem,and don't know how to solve it?
Could you help me?Thanks alot!

daniel86, 2017-08-22:
@munu8 remove devel and build. remove ~/.gradle, source bashrc of ROS distribution.

keerthanamanivannan, 2019-03-06:
Hi @daniel86, I tried doing these: 
`catkin clean`, 
then `rm -rf ~/.gradle/` and 
then `source ~/.bashrc` (which sources `/opt/ros/kinetic/setup.bash`),

but I still get this guy:
```Could not find any matches for org.knowrob.knowrob_common:knowrob_common:[0.1,) as no versions of org.knowrob.knowrob_common:knowrob_common are available.```

keerthanamanivannan, 2019-03-08:
@daniel86 ping.

asilx, 2019-03-10:
@keerthanamanivannan  did you install rosjava from source by any chance? I experienced similar errors in the setups where rosjava is built from source and there were issues with overlaying catkin workspaces

keerthanamanivannan, 2019-03-13:
@asilx No, AFAIK. I did not install rosjava from source.