knowrob/knowrob #111: Update install tutorial, add required dependencies

andreihaidu, 2016-07-27:
List of dependencies which I needed:

from [code_iai](https://github.com/code-iai):
* iai_cad_tools
* iai_common_msgs

mjpeg_server:
http://wiki.ros.org/mjpeg_server

rosbridge_server:
apt-get install ros-indigo-rosbridge-server

json:
apt-get install libjson-glib-dev

fairlight1337, 2016-07-27:
I can confirm that

andreihaidu, 2016-07-27:
Update:
apt-get install swi-prolog-java

daniel86, 2016-08-08:
@andreihaidu to which install tutorial are you referring?

andreihaidu, 2016-08-08:
Not a tutorial per se, if one clones the repository and wants to build it via catkin.

JKBehrens, 2016-08-31:
I used @gheorghelisca s workspace setup (https://github.com/gheorghelisca/ros_indigo_workspaces) to install Knowrob, CRAM, semrec and so on. I documented the whole process (last week). For knowrob, I needed to do:
`sudo apt-get install ros-indigo-rosjava* ros-indigo-mjpeg-server ros-indigo-rosbridge-server swi-prolog libjson-glib-dev`

prfiredragon, 2016-08-31:
Thank you for sharing the  workspace setup link. I was searching for something like that. That help me alot.

closed 2021-02-18 by daniel86