knowrob/knowrob #117: ERROR: rdf_db:rdf_has/3: Type error: `atom' expected, found `inverse_of(rdf:type)'

prfiredragon, 2016-08-28:
hi.I have ros-jade on ubuntu trusty.
First for compile I have commented the line 124 on /home/mike/catkin_ws/src/stacks/knowrob/knowrob_vis/knowrob_vis/src/main/java/org/knowrob/vis/TaskTreeVisualization.java because don't let me compile. now I am tring to test the Visualizing objects, and give me that error. I run this
rosrun rosprolog rosprolog knowrob_basics_tutorial
?- owl_parse('package://knowrob_map_data/owl/ccrl2_semantic_map.owl').
?- owl_parse('package://knowrob_basics_tutorial/owl/ccrl2_map_objects.owl').

on another terminal run roslaunch rosbridge_server rosbridge_websocket.launch then

on the previous terminal ?- visualisation_server.
Open your browser at http://localhost:1111
 
?- owl_individual_of(Map, knowrob:'SemanticEnvironmentMap'), marker_update(object(Map)). 
and here I get the error.

?- visualisation_server.
true.

?- 2016-08-28 14:01:15.935:INFO:oejs.Server:jetty-8.1.15.v20140411
2016-08-28 14:01:16.131:INFO:oejs.AbstractConnector:Started SelectChannelConnector@0.0.0.0:1111
owl_individual_of(Map, knowrob:'SemanticEnvironmentMap')   , marker_update(object(Map)).
ERROR: rdf_db:rdf_has/3: Type error: `atom' expected, found `inverse_of(rdf:type)'
   Exception: (9) util:time_term(knowrob:'timepoint_1472407407.639544', _G89190) ?

prfiredragon, 2016-08-29:
I uninstall ros-jade and install ros-indigo. Have the same problems. The same error to compile and the same error trying to visualize.

daniel86, 2016-09-15:
Hi @prfiredragon!
Sorry for the delayed reply, I had some busy weeks at the institute.
Unfortunately I am not able to reproduce this bug.
I tried it with _master_ and the more recent _robohow_ branch.

-  I haven't tested KnowRob with ros-jade yet, this could be related to your problem,
it should be safe to use indigo.
- Which branch did you used for testing?
- Could you please [trace](http://www.swi-prolog.org/pldoc/man?section=debugger) the error? Since I can't reproduce it, I need more input.

Note that the compilation error about TaskTreeVisualization might be related to ROS messages not being updated. This is caused by how knowrob packages include generated messages in the build process. Usually it can be fixed by removing generated message files in knowrob packages.

HelenHarman, 2016-12-05:
Hi,
I also get a similar error when running: sub_component(baxter:'baxter_baxter1', Sub). (Following the instructions at http://www.knowrob.org/doc/srdl2_tutorial)

I am using:
- SWI-Prolog version 7.2.3 for amd64
- Ubuntu 16.04
- Installed KnowRob using: http://www.knowrob.org/installation/catkin
- ROS kinetic

Appears to be issue with ComponentComposition

[image]

daniel86, 2017-03-01:
@HelenHarman Unable to reproduce with indigo:

```
$ rosrun rosprolog rosprolog knowrob_srdl

?- owl_parse('package://knowrob_srdl/owl/PR2.owl').
% Parsed "PR2.owl" in 0.05 sec; 4,399 triples
true.

?- owl_parse('package://knowrob_srdl/owl/baxter.owl').
% Parsed "baxter.owl" in 0.03 sec; 1,841 triples
true.

?- register_ros_package(knowrob_actions).
true.

?- owl_parse('package://knowrob_actions/owl/serve_drink.owl').
% Parsed "serve_drink.owl" in 0.01 sec; 465 triples
true.

?- sub_component(baxter:'baxter_baxter1', Sub).
Sub = baxter:'LeftArmController' ;
Sub = baxter:'RightArmController' ;
Sub = baxter:baxter_base ;
Sub = baxter:baxter_collision_head_1 ;
Sub = baxter:baxter_collision_head_link_1 ;
Sub = baxter:baxter_collision_head_2 
```

My guess is that it's about missing kinetic support (https://github.com/knowrob/knowrob/issues/118)
I will look into it this week

daniel86, 2017-03-06:
@HelenHarman I was able to trace the error to `owl_individual_from_range` (which is called internally when you call `sub_component`). basically this code:

    owl_individual_from_range(Resource, Class) :-
	    nonvar(Resource), !,
	    rdf_has(_, P, Resource),
	    rdf_has(P, rdfs:range, Class), !.

When the error occurs the first `rdf_has` binds `P` to 'inverse_of(...)' and the second call then crashes because only atoms (or namespaced atoms) are allowed in the first argument of `rdf_has`. A quick fix would be to ensure atoms in the call:

    owl_individual_from_range(Resource, Class) :-
	    nonvar(Resource), !,
	    rdf_has(_, P, Resource), atom(P),
	    rdf_has(P, rdfs:range, Class), !.

I need to think a bit about why this happens and what the proper fix is. Expect a fix coming soon!

closed 2017-03-08 by daniel86

HelenHarman, 2017-03-08:
Thank you for the fix.
Helen