knowrob/knowrob #118: KnowRob with ROS Kinetic and Ubuntu 16.04

airballking, 2016-09-20:
I was trying to use KnowRob with ROS Kinetic and Ubuntu 16.04. Basically, it did not compile because ```rosjava``` is not released for ROS Kinetic. 

What are our plans going forward? Shall ```kinetic``` be supported in the short-term future? If yes, what steps do need to take?

daniel86, 2016-09-20:
Hi @airballking , thanks for sharing this!
Yes, support for Kinetic would be nice, but I don't have planned yet when I will work on this subject.
There is also an open rosjava ticket about kinetic support: https://github.com/rosjava/rosjava_core/issues/222

I have never looked into rosjava internals, so it might require some effort to get into it if we need to fix it ourselves.

airballking, 2016-09-20:
@daniel86 Thanks for the quick answer, and the pointer to the rosjava issue. Then, let's see what they say!

daniel86, 2017-03-01:
`rosjava` is now avaiable for kinetic. I will try to get KnowRob running with kinetic this week

missxa, 2017-03-04:
@daniel86 did it work for you with ROS kinetic?

When I try to run json_prolog_node I keep getting an unsatisfiedLinkError 

```
java.lang.UnsatisfiedLinkError: jpl.fli.Prolog.get_default_init_args()[Ljava/lang/String;
	at jpl.fli.Prolog.get_default_init_args(Native Method)
	at jpl.JPL.getDefaultInitArgs(JPL.java:133)
```
Any suggestions how can I fix it?

daniel86, 2017-03-06:
@missxa looks like the Prolog/JPL shared library was not found by Java, did you install `swi-prolog-java`? You could try to extend `LD_LIBRARY_PATH` with the directory containing libjpl.so.

I have still issues with build files, and will continue looking into this problem now...

daniel86, 2017-03-06:
@missxa Got KnowRob working with kinetic, there is now a new [branch](https://github.com/knowrob/knowrob/tree/kinetic) with updated build files. I will try to keep it in synch with master (master will be still used for the indigo version until the IAI switches to kinetic)

daniel86, 2017-03-06:
I'll leave this open until there is version in Ubuntu repositories

missxa, 2017-03-06:
@daniel86 I do have `swi-prolog-java` and the `LD_LIBRARY_PATH` contains the directory with `libjpl.so`, but it still doesn't see the library. 
I'm using `java-8-openjdk` could this be causing a problem?

daniel86, 2017-03-06:
@missxa Might not be the case, but I have only tried the Prolog interpreter and not the json_prolog node today. I am blocked tomorrow but will test the node on Wednesday, also using Java 8. In the past, this error popped up and was related to local issues with jpl/swi-prolog installation. You can also check against the [Dockerfile](https://raw.githubusercontent.com/knowrob/docker/master/kinetic/kinetic-swi/Dockerfile) I created for installing KnowRob on a fresh Ubuntu 16.04

missxa, 2017-03-07:
It looks like I fixed the problem. you're using `jpl-3.1.4-alpha` in Knowrob and I had `jpl-7.0.1` locally. After minor adjustments in the source it works with `JPL-7` now.

missxa, 2017-03-08:
@daniel86 It looks like I fixed the problem. you're using `jpl-3.1.4-alpha` in Knowrob and I had `jpl-7.0.1` locally. After minor adjustments in the source it works with `JPL-7` for me now.

daniel86, 2017-03-08:
@missxa Glad that you found the problem, and good to know. What kinds of minor adjustments?
I was running in a segmentation fault caused by JPL bug recently and would like to try recent JPL version.

missxa, 2017-03-08:
@daniel86 I changed dependencies in gradle and there was a small conflict with Hashtable and HashMap. From what I can tell, they abandoned Hashtables in `JPL7`. 

I pushed edited json_prolog source to my repo [here]( https://github.com/missxa/json_prolog ).

bbferka, 2017-12-23:
@daniel86 I just upgraded to xenial and kinetic and was experiencing the same issues as **missxa** on both my desktop and laptop; (tried both master and kinetic branches); I ended up using his edited json_prolog which workde. Any ideas?

asilx, 2017-12-23:
@bbferka you can use kinect fork. It solves the issue without tweaking

daniel86, 2017-12-23:
@bbferka I guess it's about the version mess in gradle files.
Actually I am waiting on a patch from Jan-Hendrik about this, he was running into several issues before the EASE symposium. Wondering why he still has not submitted the patch.

bbferka, 2018-01-04:
@asilx I was not up to date on my kinetic branch before 
@daniel86 yes I could still have problems with other parts. I hope @jworch will send a PR soon :)

daniel86, 2018-01-18:
build files fixed in kinetic branch, I hope this solves most issues.
Please reopen if you still have issues.

closed 2018-01-18 by daniel86