knowrob/knowrob #131: Fix the Installation Tutorial on knowrob.org

sasjonge, 2016-12-02:
The installation tutorial at [http://knowrob.org/installation/catkin](url) doesn't work, if you try to install knowrob from scratch.

`/home/sascha/test_ws/src/stacks/knowrob/knowrob_vis/knowrob_vis/src/main/java/org/knowrob/vis/TaskTreeVisualization.java:124: error: cannot find symbol
			t.setTip("ID: " + tasks[i][1] + ", Goal: " + tasks[i][0] +
			 ^
  symbol:   method setTip(String)
  location: variable t of type Task
1 error
1 warning
`
Adding iai_common_msgs to the .rosintall-files seems to fix the problem (and the tutorial).

munu8, 2017-02-27:
I also meet the problem when I catkin_make the workspace.The branch of knowrob what I use is the master version. How can I solve the problem?Should I use the branch of indigo?Can you describe the soiution "Adding iai_common_msgs to the .rosintall-files seems to fix the problem (and the tutorial)." in detail.Thanks a lot!

sasjonge, 2017-02-27:
I'm using the master version. I solved the problem for me by adding

`- git:
    local-name: iai_common_msgs
    uri: https://github.com/code-iai/iai_common_msgs.git`

to the end of the .rosinstall (should be at <YourWorkspace>/src/.rosinstall. Then you need to call `wstool update` (Like in the tutorial). After this you should be able to build the workspace with `catkin_make`.

munu8, 2017-03-01:
I tried the method,and found it worked.Thanks a lot!
        But I meet a new problem.I want to use the browser-based visualization canvas.I launch the rosbridge websocket server in another terminal using the following command:
        roslaunch rosbridge_server rosbridge_websocket.launch
        Then I input the website http://localhost:1111/ or the website http://localhost:1111/robohow.html to the browser,but I find I can not found the both wesites.
        I just install the KnowRob in the /home/admin6/catkin_ws/src/stacks,without the resourse of knowrob_addons,should I download the knowrob_addons-master.zip from the github website,and catkin_make the resource in the folder  /home/admin6/catkin_ws?
        Or this problem is due to other reasons?I can not make sure the website http://localhost:1111/ is correct,should I modify the client address or the port?
        Could you help me?Thank you very much!

daniel86, 2017-03-01:
Hi @munu8 the current state of the web vis in the `knowrob_vis` is unfortunately a bit messy.
Please note that there is a submodule repository that includes some JS files for the browser.

The reason that this is not maintained well is that we use a docker-based application for web visualization since some time (called openEASE), you can find some installation instructions here: http://knowrob.org/doc/docker

Also note that KnowRob uses an extended marker visualization API. Thus, vis messages generated by KnowRob should also show up in rviz!

daniel86, 2017-03-01:
About the `iai_common_msgs`: It is actually listed as compile time dependency in build files as far as I can see.
Anyway, I've just added it to all the rosinstall files

closed 2017-03-01 by daniel86

munu8, 2017-03-01:
Thanks a lot!