knowrob/knowrob #136: Could not resolve all dependencies for configuration

munu8, 2017-03-10:
Hello everyone!
When I install the knowrob_addons,I meet a difficult problem.
I input the commands in the terminal:
cd ~/catkin_ws
catkin_make
Then I find the errors in the terminal:
.........................................................................................................................................................
FAILURE: Build failed with an exception.

    What went wrong:
    Could not resolve all dependencies for configuration ':knowrob_meshes:compile'.

    Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
    Required by:
    org.knowrob.knowrob_meshes:knowrob_meshes:1.0.0

    Try:
    Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
    make[2]: *** [stacks/knowrob_addons/knowrob_meshes/CMakeFiles/gradle-knowrob_meshes] error 1
    make[1]: *** [stacks/knowrob_addons/knowrob_meshes/CMakeFiles/gradle-knowrob_meshes.dir/all] error 2
    make[1]: ....

FAILURE: Build failed with an exception.

    What went wrong:
    Could not resolve all dependencies for configuration ':knowrob_memory_graph:compile'.

    Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
    Required by:
    org.knowrob.knowrob_memory_graph:knowrob_memory_graph:0.1.0

    Try:
    Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

FAILURE: Build failed with an exception.

    What went wrong:
    Could not resolve all dependencies for configuration ':knowrob_plan_summary:compile'.

    Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
    Required by:
    org.knowrob.knowrob_plan_summary:knowrob_plan_summary:0.1.0

    Try:
    Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
    make[2]: *** [stacks/knowrob_addons/knowrob_memory_graph/CMakeFiles/gradle-knowrob_memory_graph] error 1
    make[1]: *** [stacks/knowrob_addons/knowrob_memory_graph/CMakeFiles/gradle-knowrob_memory_graph.dir/all] error 2
    make[2]: *** [stacks/knowrob_addons/knowrob_plan_summary/CMakeFiles/gradle-knowrob_plan_summary] error 1
    make[1]: *** [stacks/knowrob_addons/knowrob_plan_summary/CMakeFiles/gradle-knowrob_plan_summary.dir/all] error 2

FAILURE: Build failed with an exception.

    What went wrong:
    Could not resolve all dependencies for configuration ':knowrob_chemlab:compile'.

    Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
    Required by:
    org.knowrob.knowrob_chemlab:knowrob_chemlab:1.0.0
    Could not find any version that matches org.knowrob.knowrob_sim_games:knowrob_sim_games:[0.1,).
    Required by:
    org.knowrob.knowrob_chemlab:knowrob_chemlab:1.0.0

    Try:
    Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
    make[2]: *** [stacks/knowrob_addons/knowrob_chemlab/CMakeFiles/gradle-knowrob_chemlab] error 1
    make[1]: *** [stacks/knowrob_addons/knowrob_chemlab/CMakeFiles/gradle-knowrob_chemlab.dir/all] error 2

FAILURE: Build failed with an exception.

    What went wrong:
    Could not resolve all dependencies for configuration ':knowrob_openease:compile'.

    Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
    Required by:
    org.knowrob.knowrob_openease:knowrob_openease:1.0.0

    Try:
    Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
    make[2]: *** [stacks/knowrob_addons/knowrob_openease/CMakeFiles/gradle-knowrob_openease] error 1
    make[1]: *** [stacks/knowrob_addons/knowrob_openease/CMakeFiles/gradle-knowrob_openease.dir/all] error 2
    make: *** [all] error 2
    Invoking "make -j8 -l8" failed
    ....................................................................................................................................................

I find the package knowrob_cram is located in the folder knowrob_addons,and the package knowrob_meshes, knowrob_memory_graph, knowrob_plan_summary, knowrob_chemlab, knowrob_openease are also located in the folder knowrob_addons.In addition,these packages are all depend on the package knowrob_cram.
As a result, when I catkin_make the workspace,the system could not resolve all dependencies for configuration?

Which one can help me solve the problem?
Thanks a lot!

asilx, 2017-03-10:
Hi,

Please don't replicate the issues you have already opened. 

Regarding to your problem. Which ROS distro do you use? Did you already install rosjava?

Greetings,
Asil

munu8, 2017-03-15:
The version of ROS which I use is indigo,and I have already installed rosjava.
I find the information of errors in the terminal:
..................................................................................................................................
FAILURE: Build failed with an exception.
* What went wrong:
Could not resolve all dependencies for configuration ':runtime'.
> Could not find org.ros.rosjava_messages:designator_integration_msgs:0.0.6.
 Required by:
      org.ros.rosjava_messages:iai_robosherlock_msgs:0.0.6

FAILURE: Build failed with an exception.
* What went wrong:
Could not resolve all dependencies for configuration ':knowrob_cad_parser:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_cad_parser:knowrob_cad_parser:1.0.0

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':knowrob_srdl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_srdl:knowrob_srdl:0.1.0

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all dependencies for configuration ':json_prolog:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.json_prolog:json_prolog:0.1.0

FAILURE: Build failed with an exception.
* What went wrong:
Could not resolve all dependencies for configuration ':semantic_map_to_owl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
 Required by:
      org.knowrob.semantic_map_to_owl:semantic_map_to_owl:1.0.0
.......................................................................................................................................................
Could you help me?Thanks a lot!

munu8, 2017-03-15:
@asilx ,Could you help me?Thanks a lot!

daniel86, 2017-03-15:
again a duplicate: Duplicate of: #137  :/ closing this one

closed 2017-03-15 by daniel86