knowrob/knowrob #137: The problem with knowrob_cram when compile the knowrob_addons package

munu8, 2017-03-15:
Hello,every!
     I meet a difficult problem when I compile the **knowrob_addons** package.
     The version of ROS which I use is indigo,and I have already install rosjava both from source and debian.
     I find the **CMakeLists** of **knowrob_chemlab,knowrob_memory_graph,knowrob_mashes,
knowrob_openease** all consist of "**find_package(catkin REQUIRED knowrob_cram)**",
but when I compile the knowrob_addons,I find the information in the terminal:
...............................................................................................................................................

Could not resolve all dependencies for configuration ':knowrob_meshes:compile'.
Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
Required by:
org.knowrob.knowrob_meshes:knowrob_meshes:1.0.0

Could not resolve all dependencies for configuration ':knowrob_memory_graph:compile'.
Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
Required by:
org.knowrob.knowrob_memory_graph:knowrob_memory_graph:0.1.0

Could not resolve all dependencies for configuration ':knowrob_plan_summary:compile'.
Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
Required by:
org.knowrob.knowrob_plan_summary:knowrob_plan_summary:0.1.0

Could not resolve all dependencies for configuration ':knowrob_chemlab:compile'.
Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
Required by:
org.knowrob.knowrob_chemlab:knowrob_chemlab:1.0.0

Could not find any version that matches org.knowrob.knowrob_sim_games:knowrob_sim_games:[0.1,).
Required by:
org.knowrob.knowrob_chemlab:knowrob_chemlab:1.0.0

Could not resolve all dependencies for configuration ':knowrob_openease:compile'.
Could not find any version that matches org.knowrob.knowrob_cram:knowrob_cram:[0.1,).
Required by:
org.knowrob.knowrob_openease:knowrob_openease:1.0.0
.............................................................................................................................................................

 I follow the inatsll tutorials in the website http://cram-project.org/installation.
I guess whether I should modify the CMakeLists of knowrob_chemlab,knowrob_memory_graph,knowrob_mashes,
knowrob_openease package,because they all consist of "find_package(catkin REQUIRED knowrob_cram)",and the system can not find the package **knowrob_cram**?
Could you help me?Thanks a lot!

daniel86, 2017-03-15:
Hi,

You don't have to install CRAM to get `knowrob_addons` to compile. `knowrob_cram` is part of the [repo](https://github.com/knowrob/knowrob_addons/tree/master/knowrob_cram).

What you get are basically Java build errors because gradle is not able to find JAR file for `knowrob_cram` in local maven repository. In fact this JAR should be generated by the build command, and also before if these packages are correctly listed as dependencies. I am not an expert with that but to me the buildfiles look ok.

daniel86, 2017-03-28:
Closing, since no response for about 2 weeks. Please re-open if you still have issues.

closed 2017-03-28 by daniel86