knowrob/knowrob #147: rdf_unique_id generates same sequence of ids in each run

asilx, 2017-07-27:
I am running experiments which are logged using json_prolog. In every run, similar sequence of unique ids are generated. This is a problem when different episodic memories are loaded at the same time.

For instance:
Run1
<knowrob:CRAMAchieve rdf:about="&knowrob;CRAMAchieve_OrUIzvEY">
  <knowrob:subAction rdf:resource="&knowrob;BaseMovement_VeMDiXVm"/>
  <knowrob:subAction rdf:resource="&knowrob;BaseMovement_ZxKuHwbZ"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMAcheive_bYmVjIRg"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMPerceive_FWcHvbLr"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMPerceive_kpdCJilp"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMPerceive_tDjYwuhx"/>
  <knowrob:endTime rdf:resource="&knowrob;timepoint_1500191283.7614863"/>
  <knowrob:startTime rdf:resource="&knowrob;timepoint_1500191259.265239"/>
  <knowrob:taskContext rdf:datatype="&xsd;string">PerceiveAndOpenFridgeDoor</knowrob:taskContext>
  <knowrob:taskSuccess rdf:datatype="&xsd;boolean">true</knowrob:taskSuccess>
  <rdf:type rdf:resource="&owl;NamedIndividual"/>
</knowrob:CRAMAchieve>

Run2
<knowrob:CRAMAchieve rdf:about="&knowrob;CRAMAchieve_OrUIzvEY">
  <knowrob:subAction rdf:resource="&knowrob;BaseMovement_VeMDiXVm"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMAcheive_ZxKuHwbZ"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMPerceive_kpdCJilp"/>
  <knowrob:subAction rdf:resource="&knowrob;CRAMPerceive_tDjYwuhx"/>
  <knowrob:endTime rdf:resource="&knowrob;timepoint_1500191851.6311207"/>
  <knowrob:startTime rdf:resource="&knowrob;timepoint_1500191832.6944838"/>
  <knowrob:taskContext rdf:datatype="&xsd;string">PerceiveAndOpenFridgeDoor</knowrob:taskContext>
  <knowrob:taskSuccess rdf:datatype="&xsd;boolean">true</knowrob:taskSuccess>
  <rdf:type rdf:resource="&owl;NamedIndividual"/>
</knowrob:CRAMAchieve>

daniel86, 2017-07-27:
Hi Asil! Fixed already in kinetic branch!
(seems to only happen with kinetic / 16.04)

Here is the fix: https://github.com/knowrob/knowrob/commit/526dff4ba030aed9368e6193f09a679ca1929f0c

asilx, 2017-07-27:
@daniel86  oh that is magical :+1: thanks a lot

asilx, 2017-07-27:
same patch has been applied to master branch.

daniel86, 2017-07-27:
oh ok so it was this one :) I wasn't sure anymore and had no time so I removed my comment. Glad that it's fixed :)
Are you sure the patch is also ok for Ubuntu 14.04? If not I would only apply to kinetic branch

asilx, 2017-07-28:
Yesterday I was happy about fastfix at the end of the day but you are right. doesnt work for 14.04 and indigo

asilx, 2017-07-29:
https://github.com/knowrob/knowrob/commit/db97547e14bfe0f5f630009e60cd5b9c3774d97c solves the issue

closed 2017-07-29 by asilx