knowrob/knowrob #150: JSON encoding bug in json_prolog

daniel86, 2017-08-14:
?- X='{\"X\": 847230894.4536E4}'.
X = [
 X = 8472309000000
]

ichumuh, 2017-08-14:
The loss in precision comes from a bug in json-lib:2.4. There exists a 6 year old bug report...
[https://sourceforge.net/p/json-lib/bugs/116/](https://sourceforge.net/p/json-lib/bugs/116/)
As mentioned in the comments the precision bug can be fixed by compiling against version 2.3.
I have tested it and it seems to be working.

However this does not change the fact that the result is a hashmap instead of a string.
The result is a hashmap because in JSONQuery.java encodeResult calls the put method from JSONObject which parses strings that have a json format.

daniel86, 2017-08-15:
From JSONObject documentation:

- An extra , (comma) may appear just before the closing brace.
- Strings may be quoted with ' (single quote).
- Strings do not need to be quoted at all if they do not begin with a quote or single quote, and if they do not contain leading or trailing spaces, and if they do not contain any of these characters: { } [ ] / \ : , = ; # and if they do not look like numbers and if they are not the reserved words true, false, or null.
- Keys can be followed by = or => as well as by :.
- Values can be followed by ; (semicolon) as well as by , (comma).
- Numbers may have the 0- (octal) or 0x- (hex) prefix.
- Comments written in the slashshlash, slashstar, and hash conventions will be ignored.

Wrapping the value seems to fix the hashmap result bug without introducing new issues as far as I can see.

daniel86, 2017-08-15:
Fixed by #151

closed 2017-08-15 by daniel86