knowrob/knowrob #158: Installing KnowRob on ROS indigo

wliu88, 2017-08-29:
I have been trying to install knowrob on Ubuntu 14.04 with ROS indigo. I followed the [catkin knowrob install tutorial](http://www.knowrob.org/installation/catkin). I made sure `ros-indigo-rosjava` is properly installed. I did the following:
```
> mkdir -p ~/knowrob_ws/src
> cd ~/knowrob_ws/src
> wstool init
> wstool merge https://raw.github.com/knowrob/knowrob/master/rosinstall/knowrob-base.rosinstall
> wstool update
> source /opt/ros/indigo/setup.bash
> rosdep update
> rosdep install --ignore-src --from-paths .
> cd ~/knowrob_ws
> catkin_make
``` 
I got several errors compiling the package. I removed the `~/.gradle` directory and also the `build` and `devel` in my workspace and recompiled everything. I still got the following messages:

```
FAILURE: Build failed with an exception.

What went wrong:
Could not resolve all dependencies for configuration ':knowrob_tutorial_msgs:runtime'.
> Could not find org.apache.commons:com.springsource.org.apache.commons.codec:1.3.0.
  Required by:
      org.knowrob.tutorials:knowrob_tutorial_msgs:1.0.0 > org.ros.rosjava_bootstrap:message_generation:0.3.0
> Could not find org.apache.commons:com.springsource.org.apache.commons.io:1.4.0.
  Required by:
      org.knowrob.tutorials:knowrob_tutorial_msgs:1.0.0 > org.ros.rosjava_bootstrap:message_generation:0.3.0
> Could not find org.apache.commons:com.springsource.org.apache.commons.lang:2.4.0.
  Required by:
      org.knowrob.tutorials:knowrob_tutorial_msgs:1.0.0 > org.ros.rosjava_bootstrap:message_generation:0.3.0

Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob_tutorials/knowrob_perception_tutorial/CMakeFiles/gradle-knowrob_perception_tutorial] Error 1
make[1]: *** [knowrob_tutorials/knowrob_perception_tutorial/CMakeFiles/gradle-knowrob_perception_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

FAILURE: Build failed with an exception.

What went wrong:
Could not resolve all dependencies for configuration ':knowrob_srdl:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Required by:
      org.knowrob.knowrob_srdl:knowrob_srdl:0.1.0

Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl] Error 1
make[1]: *** [knowrob/knowrob_srdl/CMakeFiles/gradle-knowrob_srdl.dir/all] Error 2

FAILURE: Build failed with an exception.

What went wrong:
Could not resolve all dependencies for configuration ':json_prolog:compile'.
> Could not find any version that matches org.knowrob.knowrob_common:knowrob_common:[0.1,).
  Searched in the following locations:
      file:/home/weiyu/knowrob_ws/devel/share/maven/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      file:/home/weiyu/knowrob_ws/devel/share/maven/org/knowrob/knowrob_common/knowrob_common/
      file:/opt/ros/indigo/share/maven/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      file:/opt/ros/indigo/share/maven/org/knowrob/knowrob_common/knowrob_common/
      https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      https://github.com/rosjava/rosjava_mvn_repo/raw/master/org/knowrob/knowrob_common/knowrob_common/
      file:/home/weiyu/.m2/repository/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      file:/home/weiyu/.m2/repository/org/knowrob/knowrob_common/knowrob_common/
      http://repository.springsource.com/maven/bundles/release/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      http://repository.springsource.com/maven/bundles/release/org/knowrob/knowrob_common/knowrob_common/
      http://repository.springsource.com/maven/bundles/external/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      http://repository.springsource.com/maven/bundles/external/org/knowrob/knowrob_common/knowrob_common/
      https://repo1.maven.org/maven2/org/knowrob/knowrob_common/knowrob_common/maven-metadata.xml
      https://repo1.maven.org/maven2/org/knowrob/knowrob_common/knowrob_common/
  Required by:
      org.knowrob.json_prolog:json_prolog:0.1.0

Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
make[2]: *** [knowrob/json_prolog/CMakeFiles/gradle-json_prolog] Error 1
make[1]: *** [knowrob/json_prolog/CMakeFiles/gradle-json_prolog.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

Could you please take a look? Is there any dependency missing from my system? Thanks!

daniel86, 2017-09-06:
1. Make sure you have only one rosjava installation -- in the past it sometimes happened to people that they had issues when there were another installation around (e.g. from source)

2. Could also be the environment. Try not sourcing of any setup.bash in your bashrc, start a new console and source only the knowrob workspace.

wliu88, 2017-09-14:
Hi Daniel! Thanks for the response. I successfully installed KnowRob on Ubuntu 16.04 with ROS Kinectic. I found out the problem is related to the change in central Maven repository as mentioned [here](https://github.com/rosjava/rosjava_core/issues/250).

For future reference, I did the following:
1. Install Apache Maven 3.5.0 from source
2. Configured to use Java 8
3. Used `master` branch of `knowrob` repo, `indigo level` branch of `knowrob_tutorials`, and `master` branch of `iai_common_msgs`
4. Downloaded all the missing dependencies from [SpringSource Repository](http://ebr.springsource.com/repository/app/bundle) and manually installed them [this way](https://maven.apache.org/guides/mini/guide-3rd-party-jars-local.html).

I plan to go back to 14.04 and install KnowRob. I will give an update on that.

Affonso-Gui, 2017-09-21:
Was experiencing the same problem, but was able to install successfully in Ubuntu 14.04 Indigo by following wliu88's instructions.

Thanks for the help!

daniel86, 2018-01-18:
build files fixed in kinetic branch, I hope this solves most issues.
Please reopen if you still have issues.

closed 2018-01-18 by daniel86