knowrob/knowrob #174: Tutorial visualisation fails on Ubuntu 16 ROS Kinetics

XB32Z, 2017-12-13:
Hi all, 

I am new to knowrob and I installed it on Ubuntu 16, ROS Kinetic. When I run the [tutorial](http://www.knowrob.org/doc/knowrob_basics), I get an error:

`ERROR: knowrob_objects:comp_pose_at_time/3: Undefined procedure: knowrob_mongo:mng_object_pose_at_time/4`

And a bunch of Java Exceptions. I have no experience with Prolog as well so I really cannot tell more. However I checked in the knowrob_mongo pacakge and the procedure is also not defined but it proposes to use `knowrob_mongo_descriptors:mng_object_pose_at_time/4`

Thx for your help, 
Best

asilx, 2017-12-13:
Thanks for reporting the issue!

In which query, exactly, are you getting this error?

You can maybe first try to query: "register_ros_package('knowrob_mongo')." Then try again the previous one

XB32Z, 2017-12-13:
Hi, 
the query is this one `owl_individual_of(A, knowrob:'SemanticEnvironmentMap'), marker_update(object(A)).`. I noticed that the Java is issuing an error about MD5 mismatch that I think is related between different version of rosjava messages.

daniel86, 2018-02-15:
@asilx when you initially answer questions here I expect that you take care of them, also close them when they are answered.

@XB32Z MD5 is due to older Java versions are not supported by gradle anymore. Please update Java to newer version, then the MD5 error should go away.
I have recently fixed the local visualization server, thus closing now, please re-open if you still have issues.

closed 2018-02-15 by daniel86