knowrob/knowrob #196: quote bug in json_prolog

ichumuh, 2018-03-21:
Query: 
rosservice call /json_prolog/simple_query "mode: 0
id: ''
query: 'X=\"a\"'"

Result:
rosservice call /json_prolog/next_solution "id: ''" 
status: 3
solution: "{\"X\":\"'a'\"}"
------------------------------------------------------------------------
But:
rosservice call /json_prolog/simple_query "mode: 0
id: ''
query: 'X=[\"a\"]'"

Result:
rosservice call /json_prolog/next_solution "id: ''" 
status: 3
solution: "{\"X\":[\"a\"]}"
------------------------------------------------------------------------
This is inconsistent, the second result should be "{\"X\":[\"'a'\"]}", or "{\"X\":\"['a']\"}", or am I wrong?

daniel86, 2018-03-21:
I can't really follow what you are doing here, the formatting you provide is terrible.

In Prolog, atoms like 'a' don't need to be wrapped in single quotes.
Does this answer your question?

daniel86, 2019-06-18:
should be fixed now

closed 2019-06-18 by daniel86