knowrob/knowrob #209: failed to build workspace

bbferka, 2019-01-04:
Hi, 

What| Version
-|-
OS | 16.04
branch| kinetic 
Java verison | 8

Clean workspace, clean checkout of the repo, up to date Ubuntu (latest updates installed),  I am getting this error

```

[100%] Gradling tasks for knowrob_common
cd /home/ferenc/work/kr_ws/src/knowrob/knowrob_common && ROS_MAVEN_REPOSITORY=https://github.com/rosjava/rosjava_mvn_repo/raw/master /home/ferenc/work/kr_ws/build/knowrob_common/catkin_generated/env_cached.sh /home/ferenc/work/kr_ws/src/knowrob/knowrob_common/gradlew -q installApp publishMavenJavaPublicationToMavenRepository writeClasspath

FAILURE: Build failed with an exception.

* Where:
Script 'https://github.com/rosjava/rosjava_bootstrap/raw/kinetic/buildscript.gradle' line: 44

* What went wrong:
A problem occurred evaluating script.
> Could not find method google() for arguments [] on repository container.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
```

Any ideas? It seems to be some gradle issue; the one with rosjava is too old or something; Did anyone else have this error?

daniel86, 2019-01-04:
could be that they have changed the build file upstream.
at least the google() statement looks new to me, and I don't what it is supposed to do.

bbferka, 2019-01-04:
Thx @daniel86  for the quick reply ....it is new...they changed things 9 dayst ago and released only part of it ...rosjava is a piece of !@#...pls let me know if you ever want to rewrite everything in C++, I volunteer for helping; Anyway...I had to recompile rosjava from source (!@$$).  I changed all the gradlew files and content of the gradle folders in the ros packages (is there no automatic way of doing this?); This is needed because  the new gradlew is version 4.10+ the one in knowrob packages is 2.10; This fixes the ``google()`` issue. 

Using ./gradlew build in the local folders now works, but, using catkin make I get:

```
Task 'installApp' not found in root project 'knowrob_common'. Some candidates are: 'install'.
```
any clues?

bbferka, 2019-01-04:
https://github.com/rosjava/rosjava_build_tools/pull/35

bbferka, 2019-01-16:
@yangshuo11 have you tried with my PR? (#210)

yangshuo11, 2019-01-16:
Many thanks for your quick @bbferka . I have already fixed this issue in another way. For example, in the knowrob_common package, edit the CMakeLists.txt and delete the 'installApp' in the line 'catkin_rosjava_setup'. In other packages, do it the same way. After removing 'installApp', build the knowrob_ws with catkin_make. It works fine.

yangshuo11, 2019-01-16:
Many thanks for your quick reply @bbferka . I have already fixed this issue in another way. For example, in the knowrob_common package, edit the CMakeLists.txt and delete the 'installApp' in the line 'catkin_rosjava_setup'. In other packages, do it the same way. After removing 'installApp', build the knowrob_ws with catkin_make. It works fine.

closed 2019-02-21 by daniel86