knowrob/knowrob #211: Installation problems

keerthanamanivannan, 2019-02-25:
### Description

Having issues while installing KnowRob with the `instructions` from [here](http://knowrob.org/installation/catkin)

### My environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

### Expected behaviour
Workspace containing KnowRob should be built without any issues.

### Actual behaviour
Workspace containing KnowRob is failing to build on these packages: `knowrob_common` and `tfjava`

### Backtrace or Console output

__________________________________________________________________________________________________________________

```
Errors     << knowrob_common:make /home/kmanivannan/hmi/knowrob_ws/logs/knowrob_common/build.make.000.log         
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine java version from '9-internal'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
CMakeFiles/gradle-knowrob_common.dir/build.make:57: recipe for target 'CMakeFiles/gradle-knowrob_common' failed
make[2]: Leaving directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_common'
make[2]: *** [CMakeFiles/gradle-knowrob_common] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/gradle-knowrob_common.dir/all' failed
make[1]: Leaving directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_common'
make[1]: *** [CMakeFiles/gradle-knowrob_common.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

So is this an issue with java installation? How do I go about fixing this? 
I've tried rm -rf the `~/.gradle` folder and the build and devel directories. 
I have rosjava binary installed.  

Any relevant thoughts/suggestions/tips will help.

asilx, 2019-02-26:
it seems that you have configured the default java version of your ubuntu as 9-internal. Could you reconfigure it as java8 and try again?

keerthanamanivannan, 2019-03-01:
Alright. So I changed the default java version of my system by using `sudo update-alternatives --config java` and picked java8. It seems to move past the java error. 

However, when I run rosdep install from the root of the ws, it says it cannot resolve rosdeps for `iai_cad_downloader` and `mjpeg_server` (this one seems to have reached EOL according [this link](http://wiki.ros.org/mjpeg_server)). So I git clone both these repos from source in my ws, and then I hit an error in building this package `knowrob_cad_parser`.
Getting this error:

keerthanamanivannan, 2019-03-01:
Closed the issue by mistake. My bad.

asilx, 2019-03-05:
If you are on 16.04 and ROS kinetic. Could you try to switch the kinetic branch?

keerthanamanivannan, 2019-03-05:
I did that just now, and I still get the same error: 
```make[2]: Entering directory '/home/kmanivannan/hmi/knowrob_ws/build/json_prolog'
make[2]: Nothing to be done for 'CMakeFiles/test_json_prolog.dir/build'.
make[2]: Leaving directory '/home/kmanivannan/hmi/knowrob_ws/build/json_prolog'
[100%] Built target test_json_prolog

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':json_prolog:compileClasspath'.
```

daniel86, 2019-03-05:
It might be not a good idea to use *knowrob_addons* at the moment.
Do you need it?

I will remove all these legacy packages soonish until then please add CATIN_IGNORE to any package in knowrob_addons that fails building.

keerthanamanivannan, 2019-03-05:
ah! I see. ok. I'll try that now. Thank you for responding quick!

daniel86, 2019-03-05:
Another note: it might be a good idea to use *kinetic* branch of knowrob and knowrob_addons.
rosjava randomly changes its build system, and I need to provide different build files for different ROS versions, unfortunately.
But we will try to get rid of rosjava in the long term.

keerthanamanivannan, 2019-03-05:
I see. yeah, I am using `kinetic` branch on both `knowrob` and `knowrob_addons`.  :+1: 

so the package which is failing right now is `json_prolog`, which is in `knowrob` and not `knowrob_addons`.
let me know what I can do.

daniel86, 2019-03-05:
yes, you should not CATKIN_IGNORE this.
The message you posted is not very informative. So what is the file that could not be resolved?
Can you provide more complete log? Could also help to *rm -r* the build directory.

keerthanamanivannan, 2019-03-05:
Ok, I apologize.
This is the error I'm getting when I `catkin build` my knowrob_ws.

daniel86, 2019-03-05:
I think these are only *warnings* that do not cause the build to fail.

So you were able to fix the issue? What was the problem? It would be nice if could share in case someone
 runs into the same issue.

keerthanamanivannan, 2019-03-05:
I'm sorry, I closed the issue by mistake. I would appreciate it if you can help me fix this. 
I get this error for `knowrob_cad_parser` repo before I get errors for `json_prolog`.

```______________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << knowrob_cad_parser:make /home/kmanivannan/hmi/knowrob_ws/logs/knowrob_cad_parser/build.make.000.log                                                                                                                     
make[2]: *** [CMakeFiles/gradle-knowrob_cad_parser] Error 1
make[1]: *** [CMakeFiles/gradle-knowrob_cad_parser.dir/all] Error 2
make: *** [all] Error 2
cd /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser; catkin build --get-env knowrob_cad_parser | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................................................................................................................................................................................................
Failed     << knowrob_cad_parser:make                    [ Exited with code 2 ]                                                                                                                                                       
Failed    <<< knowrob_cad_parser                         [ 8.4 seconds ]       ```

keerthanamanivannan, 2019-03-05:
`catkin build --verbose` gives me this:
```______________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << knowrob_cad_parser:make /home/kmanivannan/hmi/knowrob_ws/logs/knowrob_cad_parser/build.make.001.log                                                                                                                     
/usr/bin/cmake -H/home/kmanivannan/hmi/knowrob_ws/src/knowrob_addons/knowrob_cad_parser -B/home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser/CMakeFiles /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser/CMakeFiles/progress.marks
/usr/bin/make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser'
/usr/bin/make -f CMakeFiles/gradle-knowrob_cad_parser.dir/build.make CMakeFiles/gradle-knowrob_cad_parser.dir/depend
make[2]: Entering directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser'
cd /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kmanivannan/hmi/knowrob_ws/src/knowrob_addons/knowrob_cad_parser /home/kmanivannan/hmi/knowrob_ws/src/knowrob_addons/knowrob_cad_parser /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser/CMakeFiles/gradle-knowrob_cad_parser.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser'
/usr/bin/make -f CMakeFiles/gradle-knowrob_cad_parser.dir/build.make CMakeFiles/gradle-knowrob_cad_parser.dir/build
make[2]: Entering directory '/home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser'
[100%] Gradling tasks for knowrob_cad_parser
cd /home/kmanivannan/hmi/knowrob_ws/src/knowrob_addons/knowrob_cad_parser && ROS_MAVEN_REPOSITORY=https://github.com/rosjava/rosjava_mvn_repo/raw/master /home/kmanivannan/hmi/knowrob_ws/build/knowrob_cad_parser/catkin_generated/env_cached.sh /home/kmanivannan/hmi/knowrob_ws/src/knowrob_addons/knowrob_cad_parser/gradlew -q installDist publishMavenJavaPublicationToMavenRepository writeClasspath

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':knowrob_cad_parser:compileClasspath'.
```

daniel86, 2019-03-05:
Just put a empty file named *CATKIN_IGNORE* into the directory "src/knowrob_addons/knowrob_cad_parser", or remove knowrob_addons entirely from your workspace.

keerthanamanivannan, 2019-03-05:
oh I tried that a while ago, i `touch CATKIN_IGNORE`d in `knowrob_cad_parser` repo, but i found `json_prolog` to fail.

I removed `knowrob_addons` completely from the ws, but some of the repos from `knowrob_dev` need repos from `knowrob_addons`. so they fail too.

daniel86, 2019-03-05:
please also remove knowrob_dev

keerthanamanivannan, 2019-03-05:
yes. I did just that. but I'm getting stuck on an error which says it can't build `json_prolog` repo.
```
/usr/bin/cmake -E cmake_link_script CMakeFiles/object_positions.dir/link.txt --verbose=1
/usr/bin/c++      CMakeFiles/object_positions.dir/examples/object_positions.cpp.o  -o /home/kmanivannan/hmi/knowrob_ws/devel/.private/json_prolog/lib/json_prolog/object_positions  -L/usr/lib/swi-prolog/lib/amd64 -rdynamic /home/kmanivannan/hmi/knowrob_ws/devel/.private/json_prolog/lib/libjson_prolog.so /home/kmanivannan/hmi/knowrob_ws/devel/.private/rosprolog/lib/librosprolog.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options -ltinyxml /opt/ros/kinetic/lib/libroscpp.so -lboost_filesystem -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_system -lboost_thread -lboost_chrono -lboost_date_time -lboost_atomic -lpthread -lconsole_bridge -ljson-glib-1.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 /home/kmanivannan/hmi/knowrob_ws/devel/.private/rosprolog/lib/librosprolog.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options -ltinyxml /opt/ros/kinetic/lib/libroscpp.so -lboost_filesystem -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_c<===----------> 29% EXECUTING [2s]read -lboost_chrono -lboost_date_time -lboost_atomic -lpthread -lconsole_bridge -Wl,-rpath,/usr/lib/swi-prolog/lib/amd64:/home/kmanivannan/hmi/knowrob_ws/devel/.private/json_prolog/lib:/home/kmaniv<===----------> 29% EXECUTING [3s]ve dependencies of :json_prolog:compileClasspath
> :json_prolog:compileJava > Resolve dependencies of :json_prolog:compileClasspath > maven-metadata.xml
[100%] Built target object_positions

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':json_prolog:compileClasspath'.
```

daniel86, 2019-03-06:
The message is again not very helpful. What are the files that cannot be resolved? I assume there is more output that contains a hint to what is going wrong. Please paste the whole output at e.g. some pastebin service

keerthanamanivannan, 2019-03-06:
Ok, I copied all of the reds that the catkin build was throwing, and some more build output before that. 
[pastebin link](https://pastebin.com/CAxz9VZJ)

daniel86, 2019-03-06:
**Could not find any matches for org.knowrob.tfjava:tfjava:[0.1,) as no versions of org.knowrob.tfjava:tfjava are available.** <- this is the important message

daniel86, 2019-03-06:
https://github.com/knowrob/knowrob/issues/77

closed 2019-03-06 by daniel86