knowrob/knowrob #222: Undefined procedure: marker_vis:object_pose_at_time/3

mecisery, 2019-09-18:
Hi all,
I am new to knowrob and I installed it on Ubuntu 16, ROS Kinetic. 
When I run the following prolog command,
owl_individual_of(A, knowrob:'SemanticEnvironmentMap'), 
marker_update(object(A)).

I get an error:
ERROR: marker_vis:marker_lookup_transform/4: Undefined procedure: marker_vis:object_pose_at_time/3

Thx for your help,
Best

daniel86, 2019-10-09:
Thanks for the report, seems there is a bug in the current version.
I will fix it within the next days. Sorry for the inconvenience!

daniel86, 2019-11-09:
fixed in https://github.com/knowrob/knowrob/pull/224/commits/f570e2715d0627c079750fcaf7ad1da8f5a5de8d

The Prolog-based marker representation is no more, it was redundant with the object_state_publisher which is now used exclusively.

closed 2019-11-09 by daniel86

mecisery, 2019-11-19:
Alright. Thanks a lot for your response @daniel86
But,what is the "object_state_publisher"? a topic? How I run it?

daniel86, 2019-12-14:
@mecisery The object_state_publisher publishes on /tf and /marker_visualization_array topics based on messages KnowRob sends out on the /object_state topic.

here is a launch file: https://github.com/knowrob/knowrob/blob/master/knowrob_objects/launch/object_state_publisher.launch