knowrob/knowrob #229: Outdated tutorials

SDBonhof, 2020-02-06:
I have installed KnowRob following the tutorial [here](http://knowrob.org/installation/catkin) (I had to delete the ros-mongodb_log due to missing dependencies, but I don't think I need that anyway). 

Launching rosprolog using

    $ rosrun rosprolog rosprolog knowrob_common

launches rosprolog and shows me the prolog command line. Then, according to [the basic tutorial](http://knowrob.org/doc/knowrob_basics) I should be able to query

    ?- owl_subclass_of(A, knowrob:'FoodOrDrink').

resulting in subclasses of 'FoodOrDrink' as answers, which should look like:

     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#FoodOrDrink' ;
     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#Drink' ;
     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#Coffee-Beverage' ;
     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#InfusionDrink' ;
     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#Tea-Beverage' ;
     A = 'http://ias.cs.tum.edu/kb/knowrob.owl#Tea-Iced' 
     Yes.

However, the answer I am receiving is:

    A = knowrob:'FoodOrDrink' ;
    false.

Testing to see whether the Knowrob ontology is actually loaded using:

    ?- owl_parser:owl_file_loaded('http://knowrob.org/kb/knowrob.owl').
returns:

    true.

I looked into the file at http://knowrob.org/kb/knowrob.owl and I do find the Class:

    
    <owl:Class rdf:about="http://ias.cs.tum.edu/kb/knowrob.owl#FoodOrDrink">
    <rdfs:subClassOf rdf:resource="http://ias.cs.tum.edu/kb/knowrob.owl#FoodOrDrinkOrIngredient"/>
    </owl:Class>

The only way I can get it to work is by downloading the respective ontology that should be loaded from the [init.pl file](http://knowrob.org/kb/knowrob.owl) and query it using the full uri: 

    ?- owl_parser:owl_parse('/home/<user>/Downloads/knowrob.owl').
and then:

    ?- owl_subclass_of(A,'http://ias.cs.tum.edu/kb/knowrob.owl#FoodOrDrink').

which does result in the expected answers.
 
So, something somewhere along the line went wrong making the basic tutorials a pain to follow. Does anyone have any idea where things went wrong, and can this be fixed to make KnowRob easier to get into for new users?

daniel86, 2020-02-06:
Hello @SDBonhof 

There were recently breaking changes regarding the ontologies.
Namely that KnowRob now uses a well designed foundational ontology *DUL*.
Unfortunately, I haven't had the time to update the tutorials on knowrob.org yet.
Please have a look at the README files instead, they should be up-to-date and provide
some information about what is going on in the respective package.

I will soon also update the tutorials, but it will be quit an effort to do so,
so it will need some time.

Please let me know in case I can help otherwise.

daniel86, 2020-02-06:
btw. you can use the *onlyPythonSupport* branch of ros-mongodb_log. There is currently an issue with the CXX driver of mongo being removed from package repositories.

SDBonhof, 2020-02-06:
Thank you for your quick reply.

I understand that these things take time. Maybe you can already add a warning at the beginning of the tutorials stating that some things might not work, until you have had time to update the tutorials.

Is it possible that the parsing of ontologies from http:// files is broken due to the changes you mentioned? It feels to me like the function which parses ontologies should not be affected by which ontologies are being imported? I am no expert by any means so correct me if I'm wrong.

daniel86, 2020-02-06:
http is supposed to work, I don't think that I have touched the predicate *owl_parse* recently.
I will test it and report back here.

about the tutorials: yes I agree, and will do now. ultimately versioned tutorials are needed in the long term. There is some hope that KnowRob will have a release schedule soon, Then I would also plan to update the tutorials where required for each released version.
The first release might be close, as finally I could get rid of rosjava which has no release yet for melodic.

daniel86, 2020-02-06:
I've added a note in the doc page here: http://www.knowrob.org/doc

daniel86, 2020-02-06:
I do not have issues loading ontologies from http.
Is the http address publicly accessible? if so, please provide it, then I can try to load it.

Otherwise, is there any console output? *owl_parse* yields true or false?

SDBonhof, 2020-02-07:
Thank you for adding the note to the doc! And also thank you for your response.

I am trying to load http://knowrob.org/kb/knowrob.owl, which is the ontology which includes the 'FoodOrDrink' example from the tutorial. This ontology is still publicly accessible and I can find the 'FoodOrDrink' class, so it should be found if it is loaded and queried.

I am launching rosprolog using
`rosrun rosprolog rosprolog knowrob_common`

Which should already load the ontology according to these lines in the init.pl file in knowrob_common:
`:- owl_parser:owl_parse('http://knowrob.org/kb/knowrob.owl').`
and
`:- rdf_db:rdf_register_ns(knowrob,   'http://knowrob.org/kb/knowrob.owl#', [keep(true)]).`

Trying to query 
`?- owl_subclass_of(A, knowrob:'FoodOrDrink').`
returns:
`A = knowrob:'FoodOrDrink' ;`
`false.`

while the expected answer is:
`A = knowrob:'FoodOrDrink' ;`
`A = knowrob:'Drink' ;`
`A = knowrob:'Buttermilk' ;`
`A = knowrob:'Coffee-Beverage' ;`
`...cont...`

Manually loading from http://:
---
Parsing the ontology manually using
`?- owl_parser:owl_parse('http://knowrob.org/kb/knowrob.owl').`
returns:
`true.`

Then registering it under a new namespace (using knowrob2 as a namespace since I the namespace 'knowrob' already exists and I am not sure how that is handled):
`?- rdf_db:rdf_register_ns(knowrob2,   'http://knowrob.org/kb/knowrob.owl#', [keep(true)]).`
also returns:
`true.`

However, the query (this time using the namespace 'knowrob2') still returns the same answer.

Registering the URI mentioned in the ontology itself:
`?- rdf_db:rdf_register_ns(knowrob2,   'http://ias.cs.tum.edu/kb/knowrob.owl#', [keep(true)]).`
returns
`true.`
but doesn't solve the problem either.

Loading from file:///:
---
The other method is to download the ontology and save it locally under the name 'knowrob.owl'. Then load it using:
`?- owl_parser:owl_parse('file:///home/<user>/Downloads/knowrob.owl').`
returns:
`true.`
Then, registering 
`?- rdf_db:rdf_register_ns(knowrob2,   'http://knowrob.org/kb/knowrob.owl#', [keep(true)]).`
still doesn't work, but registering
`?- rdf_db:rdf_register_ns(knowrob2,   'http://ias.cs.tum.edu/kb/knowrob.owl#', [keep(true)]).`
does result in the expected answers following the query mentioned above.

daniel86, 2020-02-07:
KnowRob has an internal mapping from http prefixes to local paths, such that local ontologies are preferred over remote ones. So that you can have local changes, and try them out without it to http.
So in fact, you will not get the one from http in that case, but knowrob.owl from the knowrob_common package. And this does not have a class *knowrob:FoodOrDrink*.
I guess this is where the confusion is coming from.

rather try something like: `owl_subclass_of(A, dul:'Object').` to see e.g. all subclasses of *Object*.
You can also open knowrob.owl in [Protege](https://github.com/daniel86/ros-protege) to look at the class hierarchy.

SDBonhof, 2020-02-07:
So the init.pl file of knowrob_common loads the knowrob.owl file from that package. And this ontology doesn't have the _FoodOrDrink_ class. That makes sense to me.

However, I still do not understand why I can't load the http manually using the commands I mentioned.

For my use case I might not need to be able to load the http ontologies, but others might, and I am also just trying to understand how this works.

daniel86, 2020-02-07:
Ok here is the reason: KnowRob uses since always *package* URIs in OWL import statements.
Which is not how it is supposed to be with OWL files. Standard Protege won't know what to do with these import statements (thaht's why there is a fork of it in my github that allows to load these URIs)

I went a few steps towards fixing this by basically intercepting calls to owl_parse and looking up a registry of local OWL files when owl_parse is called with a HTTP URI. The idea was to get rid of package URIs in the ontologies used by KnowRob, such that they are more accessible, especially to people unfamiliar with ROS. What is still missing is (1) regular deployment to some http server, and (2) proper versioning of deployed ontologies. What made me switch back to package URIs for now is that many of the http URIs would still yield 404, and even worse, Protege does not like when this 404 is delivered wrapped in some error page (actually it tries to load the html as OWL file then).

SDBonhof, 2020-02-10:
So the call to parse from http is intercepted by owl_parse and instead looks at local files first? That makes sense with what you are saying. 

Thank you!

Thordreck, 2020-12-07:
Hi,
Are there any updates on this? I started trying knowrob out recently and it's being rather hard given that there are no clear working examples of what can and cannot be done with it. Even some of the readme's in the repo seem to be out of date. For example, in [the reasoning section](https://github.com/knowrob/knowrob/blob/master/src/reasoning/README.md) it mentions the `workflow_constituents/3` predicate, but there's no such predicate in the repo (there's a predicate called `workflow_constituent` with no 's', so I'm not sure if that's a typo).
Some of the explanations seem to be rather technical too, failing to illustrate how things can be useful (for example, the [action detection section)](https://github.com/knowrob/knowrob/blob/master/src/reasoning/README.md#action-detection).
If there's anything I can do to help I will be more than happy to do so.

daniel86, 2021-01-27:
@Thordreck thanks for your hint about being too technical in the READMEs. They should reflect the current state, however. Unfortunately, after writing the READMEs I haven't found the time to do an iteration over them.
Especially, I plan to write more documentation in the near future about how the tell/ask rules work as I am approaching a cool state for it in my local development branch: https://github.com/daniel86/knowrob/tree/aggregate
But it might still take some weeks to get everything right.
In addition, there is some seminar material created at the moment that involves practical tasks with KnowRob that we would translate into a tutorial afterwards (around March).
Best,
Daniel

Thordreck, 2021-01-28:
Hi Daniel,
Thanks a lot for the clarification, I understand these things take time. Looking forward to the new tutorials!

daniel86, 2021-02-18:
I have updated the KnowRob basics wiki page, most of the old tutorials remain marked as deprecated, however.
We will try to update some of them in the next months.

closed 2021-02-18 by daniel86

hawkina, 2021-05-28:
Any updates on this yet? The last time I tried (yesterday) it was still out of date, and the rosinstall for knowrob-basic pulls the master branch which I think is your dev branch?

daniel86, 2021-06-06:
> Any updates on this yet? The last time I tried (yesterday) it was still out of date, and the rosinstall for knowrob-basic pulls the master branch which I think is your dev branch?

well, the README files should be fine, and also the source code documentation.
Creating tutorials takes time, time I do not really have at the moment.
But maybe e.g. @sasjonge or @AbhijitVyas have worked on something

hawkina, 2021-06-07:
> well, the README files should be fine, and also the source code documentation.
> Creating tutorials takes time, time I do not really have at the moment.
> But maybe e.g. @sasjonge or @AbhijitVyas have [...]

the knowrob-base.rosinstall points to the master branch which is the current development branch, which i've heared broke things a few times for other people. 
I know you don't have time, but maybe we can ask someone else to keep an eye on this. Otherwise we spam you guys with 100 emails asking about how to do something over and over again. That might take more time then updating the tutorial/Readme :/

daniel86, 2021-06-25:
I haven't heard about such things