knowrob/knowrob #234: Intentions for ROS2 Support?

AndrewJSchoen, 2020-05-29:
Hi all, me and a group of others are were looking to use something like this package in our work, but are also in the process of transitioning away from ROS1 to ROS2, both because of the improved cross-platform functionality, and the declining support for python2. To this end, are you planning to support ROS2? I haven't seen anything in the documentation or associated webpages for KnowRob mentioning it. 

Thanks!

daniel86, 2020-05-29:
Hi @AndrewJSchoen, thanks for your message.
And glad that you consider using KnowRob for your work!

My research group has not yet switched to ROS2.
And me and other KnowRob developers have not yet looked into supporting ROS2. But I definitely would be interested in getting KnowRob running with ROS2. I am sure that it will happen, but I can not promise yet when this will be the case.

fmrico, 2021-02-03:
Hi @AndrewJSchoen , Did you try to start a version for ROS2? I am considering using it in a grade course of cognitive robotics, and I will use ROS2. In case I use it, it would be cool to have a ROS2 branch. What do you think, @daniel86 ?

daniel86, 2021-02-03:
@fmrico there has been a recent development regarding ROS2. But only concerning *rosprolog*. It is likely that someone will soon work on adding ROS2 support to it. but the funding for this has not been confirmed yet.
However, some parts of the C++ code in KnowRob in particular might also need to be modified to run with ROS2.
Do you know what it means in particular in ROS2? I assume it is not that every single interface in the API has changed, or?
So to be honest, I do not know yet what it would take to support both.

fmrico, 2021-02-03:
Hi @daniel86 

I have ported some packages to ROS2, and it can be very direct to translate pub/subs, services, and params from ROS to ROS2. The not so easy part is to make the most of the new features and resources that ROS2 provides: QoS, Lifecycle Nodes, mainly.

nitin5, 2022-12-07:
@fmrico 
Is ROS2 ported code hosted as public release somewhere?

fmrico, 2023-01-09:
Sorry, @nitin5 . I am not able to find it now :(

daniel86, 2024-09-02:
The ROS integration has been moved into the [knowrob_ros](https://github.com/knowrob/knowrob_ros) repository.
There is no support for ROS2 yet, but it is on the agenda.

closed 2024-09-02 by daniel86