knowrob/knowrob #236: Installation fails

hawkina, 2020-06-11:
Hey. 
I've followed your installation tutorial. At the step "rosdep install --ignore-src --from-paths ." it fails with the following  error-message:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mongodb_log: No definition of [python-rrdtool] for OS version [xenial]

Setup: Ubuntu 16.04, python 2.7, ROS Kinetic
I haven't found any hints on the KnowRob website or within the installation tutorial that it would be incompatible with my setup.

daniel86, 2020-06-11:
Hello!

The current master branch only supports *melodic*. There is a branch with the last snapshot of KnowRob that should run with kinetic. But I would suggest to move to melodic ;)

You could also try my fork: https://github.com/daniel86/knowrob/
As I have recently removed the dependency to mongodb_log.
It might take some more weeks until we merge into upstream.

Cheers!

hawkina, 2020-06-15:
HI
I'm still running Ubuntu 16.04 so I cannot change to melodic just yet. 
I'll try your fork and will let you know how it goes. 
The tutorial says that there are rosinstall files for different versions but there is only one which points to the master branch. It would be cool if there were multiple for the different versions or that there would be a mention about what the master branch currently is requiring. That might save a lot of hassle :)
Cheers!

hawkina, 2020-06-15:
I've just tried it out. Removing the dependency and using your fork made it pull the dependencies properly. 
Running catkin_make, I needed to install libbison-dev manually. It was complaining that libbison-1.0 is missing, although installing that didn't fix it, installing the dev version somehow did. 
It build until 100% and then dies with errors in MongoCursor.cpp:

/home/hasu/ros_workspace/knowrob_ws/src/knowrob/src/db/mongo/MongoCursor.cpp:77:53: error: ‘mongoc_collection_find_with_opts’ was not declared in this scope
       coll_(), query_, opts_, NULL /* read_prefs */ );

and it throws this error in a few other files as well. Is this related to melodic? If melodic is the only way, I might setup a vm for it after all or something...
Cheers!

daniel86, 2020-06-22:
I assume that the libmongoc version shipped with 16.04 does not fly with KnowRob, try updating it.

daniel86, 2020-07-17:
mongodb_log has been removed as dependency

closed 2020-07-17 by daniel86