knowrob/knowrob #237: How Knowrob's Action concept is defined in RDF format?

DangMinh24, 2020-06-28:
Hi all,

Could anyone please help me understand how Action is defined in Knowrob? According to what I understand so far, Knowrob defines concept of

daniel86, 2020-06-28:
Hello,

KnowRob uses DUL upper level ontology notion of Action.
Here is the OWL file: http://www.ontologydesignpatterns.org/ont/dul/DUL.owl

The comment in there is that an Actions is _an Event with at least one Agent that isParticipantIn it, and that executes a Task that typically isDefinedIn a Plan, Workflow, Project, etc._

DangMinh24, 2020-06-28:
Thanks for your kindly help @daniel86. Does that mean the procedure of defining _Action_ and _Event_ are the same in KnowRob? 

As far as I understand, _Event_ in KnowRob is defined as: 

>  Event is a dynamic situation in which the state of the world changes; each
instance is something one would say ’happens’. [...] Events should not be confused with TimeIntervals[...]

-- a class that contains at least _startTime_ and _endTime_ properties. Was the _Action_  defined in the same way?

Best,

daniel86, 2020-06-29:
That definition does not seem to be accurate as `State` is also seen as an event. There is a long comment about what an Event is defined as in DUL.owl. Please have a look there.

Basically an occurrence with participation, no interpretation is done in the event branch.

DangMinh24, 2020-06-29:
Thank you so much @daniel86! Very informative!

closed 2020-06-29 by DangMinh24