knowrob/knowrob #248: Current master branch is not ready to be used?

ipa-hsd, 2020-09-10:
My system setup is `Ubuntu 18.04` and `ROS Melodic`

I followed the installation instructions as per `README`, which [clones the `master` branch](https://github.com/knowrob/knowrob/blob/master/rosinstall/knowrob-base.rosinstall) of the repo. But it seems that many packages are missing as compared to previous branches. Once I run the command:
```
$ rosrun rosprolog rosprolog knowrob
?- owl_has(A, rdf:type, knowrob:'Drawer').
ERROR: Unknown procedure: owl_has/3 (DWIM could not correct goal)
```
Also I tried another option:
```
$ roslaunch knowrob knowrob.launch  

[INFO] [1599762534.580791]: json_wrapper service is running
[ INFO] [1599762534.665057759]: The knowledge base is starting up, that may take a few moments!
Warning: /home/hsd/projects/kogrob/monitoring/knowrob_ws/src/knowrob/src/db/tripledb.pl:410:
Warning:    Singleton variables: [S,P,Scope,Options]
% Parsed "unit.owl" in 0.19 sec; 7,134 triples
[ INFO] [1599762535.730362824]: The knowledge base has finished starting up!
[ INFO] [1599762535.737364051]: rosprolog service is running.
```
```
$ rosrun rosprolog rosprolog_commandline.py 
[INFO] [1599762546.960123]: waiting for rosprolog services
[INFO] [1599762546.968144]: rosprolog services ready
namespaces:
dc: http://purl.org/dc/elements/1.1/
dcterms: http://purl.org/dc/terms/
eor: http://dublincore.org/2000/03/13/eor#
foaf: http://xmlns.com/foaf/0.1/
owl: http://www.w3.org/2002/07/owl#
rdf: http://www.w3.org/1999/02/22-rdf-syntax-ns#
rdfs: http://www.w3.org/2000/01/rdf-schema#
serql: http://www.openrdf.org/schema/serql#
skos: http://www.w3.org/2004/02/skos/core#
void: http://rdfs.org/ns/void#
xsd: http://www.w3.org/2001/XMLSchema#
dul: http://www.ontologydesignpatterns.org/ont/dul/DUL.owl#
qudt: http://data.nasa.gov/qudt/owl/qudt#
soma: http://www.ease-crc.org/ont/SOMA.owl#
io: http://www.ontologydesignpatterns.org/ont/dul/IOLite.owl#
knowrob: http://knowrob.org/kb/knowrob.owl#
test: http://knowrob.org/kb/pancake.owl#
urdf: http://knowrob.org/kb/urdf.owl#
-----------
?- owl_subclass_of(A, knowrob:'FoodOrDrink').
Prolog query failed: error(existence_error(procedure,:(rosprolog_query,/(owl_subclass_of,2))),context(:(rosprolog_query,/(rosprolog_query,2)),_110))
```
On taking a closer look `semweb` and `thea` libraries are missing. Not sure if that is the only problem. What am I missing here? Do I need to clone one of the older branches? Or do I need to install these missing libraries manually?

ipa-hsd, 2020-09-10:
The following is working for me:
```
$ rosrun rosprolog rosprolog knowrob
?- use_module(library(semweb/rdf_db)).
true.
?- rdf_load('/home/hsd/projects/kogrob/monitoring/knowrob_ws/src/knowrob_tutorials/knowrob_basics_tutorial/owl/coffeecup-solution.owl', [format(xml)]).
% Parsed "coffeecup-solution.owl" in 0.00 sec; 22 triples
true.
?- rdf(Var1, 'http://ias.cs.tum.edu/kb/coffeecup.owl#containsDrinkableLiquid', Var2).
Var1 = http://ias.cs.tum.edu/kb/coffeecup.owl#cup0,
Var2 = http://ias.cs.tum.edu/kb/coffeecup.owl#coffee0.
```
But not sure if it was intended to be done this way.

kaviyachandran, 2020-09-11:
> ?- owl_subclass_of(A, knowrob:'FoodOrDrink').
> Prolog query failed: error(existence_error(procedure,:(rosprolog_query,/(owl_subclass_of,2))),context(:(rosprolog_query,/(rosprolog_query,2)),_110)) [...]

There was restructuring but it is complete now. "owl_subclass_of" is replaced by is_a or instance_of. Please check Readme in each folder and unit tests to get a better understanding of the usage of is_a, instance_of. There will be a new tutorial published soon.

ipa-hsd, 2020-09-11:
Thanks @kaviyachandran! That was helpful. I will close the issue for now.