knowrob/knowrob #256: urdf_load fails

seba90, 2020-10-01:
While executing

` urdf_load('http://knowrob.org/kb/PR2.owl#PR2_0', '/home/koralewski/catkin_ws/ros_cram/src/knowrob/urdf/pr2_for_unit_tests.urdf', [load_rdf])" `

I get the following error:

> error(resource_error(_124),tell(triple(_124,http://www.ontologydesignpatterns.org/ont/dul/DUL.owl#isAbout,http://knowrob.org/kb/PR2.owl#PR2_0)))

Before calling urdf_load I am able to run 
`tripledb_load('/home/koralewski/catkin_ws/ros_cram/src/knowrob/owl/robots/PR2.owl')`
successfully.

The path to the urdf is valid and the file is existing. Any ideas ? I am using the current master branch.

daniel86, 2020-10-01:
thx for the report, I'll have a look. seems something buggy with creating the InformationObject instances

sasjonge, 2020-10-07:
This bug should be fixed by this PR:

https://github.com/knowrob/knowrob/pull/257

closed 2020-10-07 by daniel86

seba90, 2020-10-13:
Sorry for the delay answer. However, I am using your PR and now I am getting the following error:

> Prolog query failed: error(type_error(text,io:DigitalResource),context(system:atomic_list_concat/2,_100358))

AbhijitVyas, 2020-10-13:
Hello

This is because Sascha's PR

Moved tripledb_load #258

might still not be accepted in Knoworb repo.

Best,
Abhijit

sasjonge, 2020-10-14:
The PR is accepted and it should work now

seba90, 2020-10-15:
Its working. Thank you.