knowrob/knowrob #269: Errors while running rosprolog

piotrpowerpalka, 2020-12-17:
I have installed knowrob on Ubuntu 18.04 according to specifications given on https://github.com/knowrob/knowrob

When I try to run rosprolog with knowrob database I've got the following errors:

rosrun rosprolog rosprolog knowrob

[ INFO] [1608200506.432788606]: The knowledge base is starting up, that may take a few moments!
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/db/mongo/tripledb/triples.pl:745:
	Wrong context: table/1 can only be used in a directive
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/db/mongo/tripledb/triples.pl:745:
	Goal (directive) failed: mng_triples:table(get_scope_document_/2)
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/db/tripledb.pl:412:
	Singleton variables: [S,P,Scope,Options]
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/__init__.pl:49:
	Unknown message: mng_error(erase_error('No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on \'localhost:27017\']'))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/__init__.pl:49:
	Goal (directive) failed: user:tripledb_init
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/model/OWL.pl:71:
	Unknown message: mng_error(erase_error('No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on \'localhost:27017\']'))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/model/OWL.pl:71:
	Goal (directive) failed: model_OWL:tripledb_load('http://www.w3.org/2002/07/owl.rdf',[graph(common),namespace(owl,'http://www.w3.org/2002/07/owl#')])
% Parsed "unit.owl" in 1.90 sec; 7,134 triples
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/model/DUL/Object.pl:41:
	Unknown message: mng_error(erase_error(No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on 'localhost:27017']))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/model/DUL/Object.pl:41:
	Goal (directive) failed: model_DUL_Object:tripledb_load(http://www.ontologydesignpatterns.org/ont/dul/DUL.owl,[namespace(dul)])
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/model/DUL/Event.pl:44:
	Unknown message: mng_error(erase_error(No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on 'localhost:27017']))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/model/DUL/Event.pl:44:
	Goal (directive) failed: model_DUL_Event:tripledb_load(http://www.ontologydesignpatterns.org/ont/dul/DUL.owl,[namespace(dul)])
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/model/__init__.pl:12:
	Unknown message: mng_error(erase_error(No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on 'localhost:27017']))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/model/__init__.pl:12:
	Goal (directive) failed: user:tripledb_load(http://www.ontologydesignpatterns.org/ont/dul/IOLite.owl,[namespace(io)])
ERROR: /home/ubuntu/catkin_ws/src/knowrob/src/model/__init__.pl:16:
	Unknown message: mng_error(erase_error(No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on 'localhost:27017']))
Warning: /home/ubuntu/catkin_ws/src/knowrob/src/model/__init__.pl:16:
	Goal (directive) failed: user:tripledb_load(http://knowrob.org/kb/knowrob.owl,[namespace(knowrob)])
...

daniel86, 2020-12-17:
Hello @piotrpowerpalka 
most of the errors are caused by mongo DB server not running on your system.
We have recently switched to storing triple data in mongo and thus it is required
to have one running.
The other error about the table does not cause serious harm. I haven't looked into this yet but my guess is that it is caused by an old SWI Prolog version as an my Arch Linux system it does not happen.

piotrpowerpalka, 2020-12-18:
Thank you a lot, it worked!
I suggest adding the note about that to the installation tutorial. I've installed the tool on the fresh Ubuntu 18.04, and below are instructions:

`  sudo  apt-get install libmongoc-dev
  sudo  apt-get install libbson-dev
  sudo  apt-get install libbson-1.0-0
  sudo apt-get install swi-prolog
  
  wget -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -
  echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu bionic/mongodb-org/4.4 multiverse" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list
  
  sudo apt-get update
  sudo apt-get install -y mongodb-org=4.4.2 mongodb-org-server=4.4.2 mongodb-org-shell=4.4.2 mongodb-org-mongos=4.4.2 mongodb-org-tools=4.4.2
  
  mkdir -p ~/catkin_ws/src
  cd ~/catkin_ws
  catkin_make
  
  source /opt/ros/kinetic/setup.bash
  source devel/setup.bash
  

cd ~/catkin_ws/src
wstool init
wstool merge https://raw.github.com/knowrob/knowrob/master/rosinstall/knowrob-base.rosinstall
wstool update

cd ~/catkin_ws
rosdep update
rosdep install --ignore-src --from-paths src/
catkin_make
  `

daniel86, 2020-12-20:
@piotrpowerpalka thanks for sharing your set-up, I'll update the installation documentation after the holidays.

fmrico, 2021-02-03:
Hi @daniel86 

I have a related error, but in my case,  mongo is running:

[image]

Any idea of this error?

Thanks!

daniel86, 2021-02-03:
Hello @fmrico 

the error about the tabling I am aware of, and it will disappear as soon as I merge in my current working branch into master.
it should however not create problems with the execution.
The reason `owl_subclass_of/3` does not find any subclasses is most likely that there are no such subclasses anymore in the ontologies that are auto-loaded by KnowRob.
I assume you took it from an example on knowrob.org? Unfortunately these are widely not up-to date with recent developments. I know it is time to update them, sorry about the inconvenience!

About the ontologies auto-loaded by KnowRob. There have been some developments over the last years.
Summarizing KnowRob has switched to a well formalized foundational ontology, the DUL ontology, and knowrob.owl has been modularized and the more general concept have been more carefully formalized such that these definitions are applicable across different disciplines (not limited to robotics).
The more general concepts are now defined in an ontology that I have called [SOMA](https://github.com/ease-crc/soma).
SOMA has also a pretty website auto-deployed in github CI: https://ease-crc.github.io/soma/
The state of the is already Ok. However, knowrob.owl is not yet in the best shape again after this effort.

Best,
Daniel

fmrico, 2021-02-03:
Thanks, @daniel86 

Then I should wait until this merge. As I wrote in #234 , I intend to port it to ROS2 because I would like to see how to integrate it with https://github.com/IntelligentRoboticsLabs/ros2_planning_system . I work with @chcorbato on some projects, and he recommended me this repo for reasoning.

Best

JackeggAdam, 2021-07-15:
Hi @daniel86 

I have a related error while running 'rosrun rosprolog rosprolog knowrob'

~$ rosrun rosprolog rosprolog knowrob
[ INFO] [1626318736.364640376]: The knowledge base is starting up, that may take a few moments!
[ INFO] [1626318736.369469965]: mng_db_name(roslog)
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:242:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:243:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:244:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:245:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:297:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:298:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/mongolog/unification.pl:299:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:40:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:46:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:66:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:84:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:94:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:105:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:132:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:140:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2
ERROR: /home/dbq/ros_ws/knowrob_ws/src/knowrob/src/lang/terms/temporal.pl:152:
	lang_query:kb_expand/2: Undefined procedure: lang_query:comma_list/2

...
any idea of this error? Thank!

daniel86, 2024-04-02:
The issue is related to an interface that has not been migrated to the C++ re-implementation of KnowRob. We won't fix remaining issues in the legacy code.

closed 2024-04-02 by daniel86