knowrob/knowrob #275: occurs/1 predicate does not work as expected

Thordreck, 2021-01-15:
Hi,
First of all thanks for you work on this repo. I'm using extensively in my own project and I find it really useful.
I started working with events recently, and I have noticed that the `occurs` predicate is not working as I expected.
I'm listing below a few examples of weird behavior I have come across. Note that each example is self-contained and it starts with a clean knowledge base.

### Time scopes do not seem to work properly.
Given the following commands:
```
?- tell(instance_of(testEvent, dul:'Event')).
true.

?- tell(occurs(testEvent) during [10.0, 50.0]).
true.
```

I would expect queries for this event to only be valid for intervals contained in the original scope - that is `[10.0, 50.0]`.
However, I'm getting the following results:
```
?- ask(occurs(testEvent)).
true.

?- ask(occurs(testEvent) during [60.0, 70.0]).
true.
```

It seems the ask rule always returns true regardless of the original scope.

### Unbound variables in ask rules do not work
I expected to be able to ask the knowledge base what events are taking place at any given interval. However, it seems it's not possible to use unbound variables together with `occurs`. Below an example:

```
?- tell(instance_of(testEvent, dul:'Event')).
true.

?- tell(occurs(testEvent) during [10.0, 50.0]).
true .

?- ask(occurs(Events)).
false.

?- ask(occurs(Events) during [10.0, 50.0]).
false.

?- ask(occurs(testEvent)).
true.
```

### Instantiating an event and calling occurs in the same query does not work

Concatenating a tell rule to create an event instance and another one calling `occurs` returns false the first time. The query has to be executed a second time for it to work. Example below:

```
?- tell(instance_of(testEvent, dul:'Event')), tell(occurs(testEvent)).
false.

?- tell(instance_of(testEvent, dul:'Event')), tell(occurs(testEvent)).
true 
```

### An exception is thrown if the event instance was created with a time scope

If the event instance was created with its own time scope, any subsequent `occurs` query would fail. Example below:

```
?- tell(instance_of(testEvent, dul:'Event') during [10.0, 50.0]).
true.

?- tell(occurs(testEvent) during [10.0, 50.0]).
ERROR: Arithmetic: `Infinity/0' is not a function
ERROR: In:
ERROR:   [38] Infinity>50.0
ERROR:   [37] temporal_scope:time_scope_max_(Infinity,50.0,50.0) at /media/thordreck/D/projects/haru/catkin_ws/src/knowrob/src/lang/scopes/temporal.pl:169

(I'm omitting the rest of the error stack).

?- tell(occurs(testEvent)).
ERROR: Arithmetic: `Infinity/0' is not a function
ERROR: In:
ERROR:   [37] Infinity>50.0
ERROR:   [36] temporal_scope:time_scope_max_(Infinity,50.0,50.0) at /media/thordreck/D/projects/haru/catkin_ws/src/knowrob/src/lang/scopes/temporal.pl:169

(I'm omitting the rest of the error stack).
```

daniel86, 2021-01-27:
Thanks for the report @Thordreck.
seems occurs is having quite some issues. I will add more unit tests covering the cases you listed

daniel86, 2024-09-02:
The predicate has been removed from mongolog in the *dev* branch.
The reasoning being the temporal operators that are part of the query language.

closed 2024-09-02 by daniel86