knowrob/knowrob #276: Current state of computables

Thordreck, 2021-01-22:
Hi,
I would appreciate if I could get a quick rundown on how `computables` are supposed to be added the knowledge base.
According to the [docs](http://www.knowrob.org/doc/defining_computables), computables are to be stored directly in the ontology as a `PrologProperty` that points to the predicate to execute. I haven't found any example in the repo that uses this so far. Given that the docs seem to be outdated, I'm not sure if that approach is still valid.
There's also a [computables module](https://github.com/knowrob/knowrob/blob/6b6dc2870f5f8a41c02250288729964fc8054d85/src/lang/computable.pl) that exposes a `computables/1` predicate. Said predicate modifies the knowledge base to add a `can_answer` and `infer` predicates in the detected input module. I tried to use it, but I have not managed to make it work. The implementation of said predicate does not look correct, I think. The asserted `infer` predicate does not have the same signature used by other reasoners that takes _Query_, _Fact_, _QueryScope_ and _FactScope_. I haven't found any predicate that uses `computables/1` in the repo either.
Could you please confirm if, in its current state, the `computables/1` predicate is working as expected. And if so, could you please provide an example of its usage? What's the correct way to add computables right now?
Thanks!

daniel86, 2021-01-27:
Hi @Thordreck!

sorry for the delay. I am currently focusing on doing another iteration over the query language of KnowRob.
Computables are still supported without restrictions in the current master.
I added some syntactic sugar to mark binary predicates as computables in `module/2` expressions.
You can see an example here: https://github.com/knowrob/knowrob/blob/master/src/reasoning/temporal/interval.pl#L5
This will register `interval_equals/2` as computable for the property `soma:simultaneous`.
I hope this works for you!

However, there might be restrictions coming to computables very soon. Namely that they need to expand into mongo DB. Then you will only be able to implement computables based on data drawn from mongo DB.

Best,
Daniel

daniel86, 2021-01-28:
I'll have a look at what's wrong with this hook. It seems no test case covers this yet, so something might broke without me noticing. I'll have a look at it in the next days.

What I am working on right now is basically to run Prolog programs in mongo aggregation queries. There were requirements imposed by projects to use proper DB backends in KnowRob, so I decided to move to mongo DB last year as I had some experience with it before. Plus we were using mongo DB before for logging ROS messages.
Doing individual queries at each term in a query was too expensive, so I was starting to look into evaluating conjunctive queries by mongo DB aggregation.
However, aggregation can only be performed if all the data is available in mongo DB.
mongo also supports custom JavaScript functions that are called in aggregation pipelines.
This could be a way to include external information into this mechanism, but I haven't had a look.
Definitely there are some performance concerns involved with doing this, but potentially it would be possible to hook a JS function executed by mongo to a ROS call.
Until now I was considering to rather look into employing components that feed data into mongo such that it becomes available for aggregation. So you could for example, have a script polling the weather forecast, storing it in mongo and then write some Prolog rules around it which will then be evaluated by mongo DB.
Does this sound like a feasible solution to you?

Thordreck, 2021-02-03:
Hi Daniel,
> I'll have a look at it in the next days.

Thanks!

> So you could for example, have a script polling the weather forecast, storing it in mongo and then write some Prolog rules around it which will then be evaluated by mongo DB.

I understand that what you described could improve the performance of queries that work on information that's constantly being published (I'm guessing a use case would be the TF's data). This, however, would be detrimental for queries that are not run at such a high-rate IMO.  It also feels like MongoDB will be less and less hidden from external users, isn't it? From your description it looks like we would need to feed information directly into the database or to embed hooks into it.

Would it be possible to still support computables with no restrictions? Some kind of "use it at your own risk" thing.
Thanks a lot for your help.

Thordreck, 2021-02-03:
Just another idea. What about introducing some other kind of concept separated from computables for the case of "fetching information from external services"? Maybe I was just trying to use the wrong tool. I'm not sure if this would help, though.

daniel86, 2021-02-03:
Well the $function command in mongo seems like a way to include procedures without restriction. And maybe there are ways to create a more user friendly wrapper around this.
You can also easily "compile-in" additional information into queries executed by mongo, but that information cannot depend on conditions evaluated within the query by mongo, however you could evaluate the conditions separately within Prolog or by doing another mongo query, then get the data, and feed the data as constants to the next query.
But that wouldn't really transparently integrate with the querying language.

As long as it is about data, I do not see a problem with storing it in mongo. It must be somewhere anyway.
Using procedures within queries is another topic, but seems with $function it should be doable.
But I didn't reach the point yet to make a decision about that, and haven't weighted different options.
For now, $function seems like something one could support in the short term.

Anyway, I haven't reached the point to make a decision about how to proceed with computables but I tend towards looking into the $function command which seems like a nice way of doing it for the moment.
In the long term, my focus will shift more towards the organization of reasoning in KnowRob which would then also include calling non formalized blackbox procedures such as computables.
However, how to properly organize such a hybrid reasoning system is not a trivial problem.
And the old way how computables have worked certainly is not the answer.

daniel86, 2021-04-13:
Computable predicates are now working again in my branch.
Here is an example: https://github.com/daniel86/knowrob/blob/aggregate/src/lang/computable.pl#L156
I will try to merge into master soon