knowrob/knowrob #281: I meet some errors when i install knowrob.

feifanrensheng, 2021-02-16:
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by "knowrob_common"
with any of the following names:

knowrob_commonConfig.cmake
knowrob_common-config.cmake

Add the installation prefix of "knowrob_common" to CMAKE_PREFIX_PATH or set
"knowrob_common_DIR" to a directory containing one of the above files. If
"knowrob_common" provides a separate development package or SDK, be sure it
has been installed.
Call Stack (most recent call first):
knowrob_addons/comp_germandeli/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/neo/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/neo/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

sasjonge, 2021-02-16:
Hi @feifanrensheng ,

knowrob_common doesn't exist anymore and was integrated into knowrob. The up to date installation tutorial can be found on https://github.com/knowrob/knowrob

The packages in knowrob_addons weren't updated to work with the newest  version of knowrob. If you need a specific package from knowrob_addons it would have to be updated. If you simply want to install knowrob then they aren't necessary.

Best, Sascha

feifanrensheng, 2021-02-17:
> Hi @feifanrensheng ,
>
> knowrob_common doesn't exist anymore and was integrated into knowrob. The up to date installation tutorial can be found on https://github.com/knowrob/knowrob
>
> The packages in [...]

But, when I learn the Tutorial: KnowRob basics, the command "  rosrun rosprolog rosprolog knowrob_common“ will be failed.

daniel86, 2021-02-18:
I have updated the KnowRob basics wiki page

closed 2021-02-18 by daniel86