knowrob/knowrob #285: Scope issues in SWRL rules

Thordreck, 2021-03-19:
Hi,
I have been getting into SWRL rules recently. Things work great for the most part, but I have run into (what I assume are) scope issues with some rules.
Consider the following rule with only one SWRL built-in in its body:

```
startsWith("hello", "hello") -> Person(?thing).
```
The resulting prolog predicate after parsing is the following:

```
swrl_instance_of(_, 'http://www.ontologydesignpatterns.org/ont/dul/DUL.owl#Person', A) :-
    ask([{atom_concat(hello, _, hello)}], A).
```
Running the following commands:

```
tell(instance_of(thingTest, owl:'Thing')), instance_of(thingTest, dul:'Plan').
```
Results in an exception thrown in the predicate `mng_triples:scope_overlaps_query`, complaining that "its arguments are not sufficiently instantiated". I investigated a bit with the debugger and it seems the problem is that the `ask` command generated in the swrl rule (`ask([{atom_concat(hello, _, hello)}], A)`) does not instantiate the scope variable `A`. This does not happen with other rules containing references to properties or classes. I'm assuming it's because the scope is correctly unified in the `ask` command in those cases.
For now I fixed it by checking the resulting scope after the calls to either `swrl_holds` or `swrl_instance_of`. If it's not ground, I set it to the universal scope. I'm adding the relevant modifications below (file swrl.pl):

```
infer( holds(S,P,O),
       holds(S,P,O), QScope,FScope) :-
  swrl_holds1(S,P,O,QScope->FScope),
  swrl_infer_scope(FScope).

infer( instance_of(S,O),
       instance_of(S,O), QScope,FScope) :-
  once((var(O);atom(O))),
  swrl_instance_of(S,O,QScope->FScope),
  swrl_infer_scope(FScope).

%%
swrl_infer_scope(FScope) :-
  nonvar(FScope).

swrl_infer_scope(FScope) :-
  var(FScope),
  universal_scope(FScope).
```
I simply added a call to `swrl_infer_scope` in the `infer` predicates to initialize `FScope` if needed.
This seems to work, but I'm not sure if it's the correct way to do it.
What do you think? Thanks!

daniel86, 2021-05-12:
Hi @Thordreck, thanks again for digging into these thing, much appreciated!

I think this is appropriate. Within a conjunction, before calling the first goal you would have nothing that restricts the scope, then each subgoal in the conjunction may further constrain the scope, basically computing the intersection with the scope before calling.

daniel86, 2021-05-14:
The problem should be fixed now. Please re-open if there are still issues.

closed 2021-05-14 by daniel86