knowrob/knowrob #286: Token parsing issues in SWRL rules

Thordreck, 2021-03-19:
I encountered a small issue with SWRL rules containing properties or classes with underscores in their names. The following rule is not correctly parsed:

```
my_class(?thing) -> Plan(?thing).
```

The problem seems to be the call to `tokenize_atom` inside the predicate `swrl_phrase` in the file `parser.pl`. I'm adding a screenshot of the debugger below. You can see how `my_class` has been split.

[image]

daniel86, 2021-04-12:
Thanks for the report! I'll have a look into this.

Thordreck, 2021-04-15:
Hi Daniel,
Thanks for taking the time to look into this. I just found two more small issues related to the SWRL rules parsing. I'm gonna report them here in this comment, rather than opening a new issue if you don't mind.

### Cannot write decimal numbers

It seems like decimal numbers are not correctly parsed. Consider the following rule:

```
hasDataValue(?thing, ?value), greaterThan(?value, 0.8) -> Person(?thing).
```

Parsing will stop at the point in the decimal number, as shown in the screenshot below (you can see how the `RuleAtom` variable appears to be incomplete):

[image]

I managed to make it work by modifying the rule to only stop at a point after a closing parenthesis. I'm not sure if it's the proper way to do it, though. I'm adding the modified predicate below:

```
swrl_file_rule_(NS,RuleTerm) -->
  blanks, string(RulesCodes), ")", blanks, ".",
  { atom_codes(RuleAtom1, RulesCodes),
    atom_concat(RuleAtom1, ')', RuleAtom), !,
    swrl_phrase(RuleTerm, RuleAtom, NS) }, !
```

### Strings literals are not properly parsed

Strings containing characters treated as token delimiters are not properly parsed. Here's a rule that it's affected by this:

```
hasDataValue(?thing, ?value), contains(?value, "type.person") -> Person(?thing). 
```

Assuming that points are properly parsed (otherwise parsing will stop at the point in `"type.person"`), you end up with the situation shown in the screenshot below, where parsing fails because the point in the middle splits the contents in two. I'm not sure how to handle this.

[image]

daniel86, 2021-05-16:
1. The first issue is indeed related to `tokenize_atom/2` splitting atoms at underscore characters. I think best would be to use a different tokenizer, or to fix the token sequence after calling it by iterating once over all tokens and creating a modified token sequence on the way
2. I cannot reproduce the second issue. Parsing floats seems to work fine here, this I have tried: `['greaterThan(?age, 16.5)', '->', 'Adult(?p)']` and the parser output seems fine.
3. Hmm I see, I guess best would be again if this would be handled better in the tokenizer, such that "type.person" only yields one token. Probably it would be ok to handle quotations in the tokenizer.

So it seems most of the issues could be solved through a better tokenizer