knowrob/knowrob #289: Notification hooks called with same event several times

Thordreck, 2021-04-02:
Hi there,
I have been working recently with the notification system KnowRob offers through the `notify:notify_hook/1` predicate. I noticed that in some cases I was receiving the exact same notification several times. I'm gonna focus on the case of generating new individuals in the knowledge base.
Let's say we create a new individual of the class `dul:'Person'`  as follows:
```
tell(instance_of(personTest, dul:'Person')).
```
After running the command above, `notify/1` will be called and, assuming there's a hook set up, it will be called with `object(personTest)` as its argument. That's all fine and dandy. The problem comes when a command that triggers the reasoning pool is run. Take the following command as an example:
```
setof(Person, ask(instance_of(Person, dul:'Person')), People).
```
This triggers `notify/1` and any registered hooks with `object(personTest)` as its argument again. This actually happens every single time the command above is executed.
I think the problem steams from the fact that the `infer1/6` predicate (in `reasoning/pool.pl`) calls `tell/2` with whatever information the reasoner has found. In the previous command, the owl reasoner would infer that `personTest` is an instance of `dul:'Person'`, thus calling `notify/1`. Also, predicates like `instance_of` are not marked as "cachable" in the reasoning pool (`is_cachable_/1` predicate).
I'm not entirely sure if this is the intended behavior, though. In my opinion a notification for individuals should be sent only once when a new individual has been created in the knowledge base the first time, not every time a reasoner finds it. What do you think?
Thanks!

daniel86, 2021-04-12:
Hi @Thordreck. yes the behavior you report seems unwanted.

I have worked on an alternative to notify, the watch predicate:
https://github.com/daniel86/knowrob/blob/aggregate/src/lang/db.pl#L455
It currently only allows to watch triples stored in mongo DB, and it is not linked to OWL reasoning.
So you could watch rdf:type and then add new facts about an instance derived
from its newly added type.
It requires to configure mongo DB though.

It is further the case that the reasoning_pool implementation was just a placeholder.
I replaced it now with something more appropriate for the future (I hope):
https://github.com/daniel86/knowrob/blob/aggregate/src/lang/query.pl#L123
The idea is that subgoals in a conjunctive query can be answered by different backends such as mongo DB, or through computables. So we map each query to a pipeline of steps where each step produces instantiations that are passed on to the next step. This is to go towards the direction of a Watson-like system for robots.

However, I haven't integrated this yet with the OWL reasoning rules we encode in Prolog.
One of the reasons is that the future of these rules is uncertain.
We might rather work on a proper interface to an established OWL reasoner instead.
But could be that it makes sense to keep some of these rules, even if they give only incomplete results
for some cases.
I think, the more common mode of operation is that the reasoner classifies the whole knowledge base
and maybe reports back results stream-based. But then you couldn't direct the reasoner to some
specific task. However, it maybe possible to perform reasoning at certain levels of abstraction and granularity,
and maybe also to create goal-directed knowledge views for reasoning.

@sasjonge this discussion might also be interesting for you

daniel86, 2021-05-14:
I have marked the report as invalid and close it now as the notify subsystem was removed as explained above.

closed 2021-05-14 by daniel86