knowrob/knowrob #292: Notify on data/object property change

Thordreck, 2021-04-15:
Would it be possible to get notifications when a object or data property is modified? Similar to how it's done right now when a new instance is created. So far I have implemented it as follows for my own personal use.

For object properties I simply added `notify(object_property(...))` in the `holds` tell rule:

```
holds(S,P,O) +>
  { is_object_property(P), ! },
  % O1 should be the bare IRI atom
  { strip_operator_(O,O1,=), atom(O1) },
  triple(S,P,string(O1)),
  notify(object_property(S, P, O)).
```

For data properties I did something similar (note the `notify(data_property(...))` at the end):

```
holds(S,P,DataTerm) +>
  { is_data_property(P), ! },
  { strip_operator_(DataTerm,DataTerm1,=) },
  % extract value and unit from given data term
  { data_term_unit_(DataTerm1,DataValue,DataUnit) },
  % infer the base type of this property (e.g. float, string, term, etc.)
  { data_base_type_(P,DataValue,BaseType) },
  %%
  { Term=..[BaseType,DataValue] },
  holds_data(S,P,unit(Term,DataUnit)),
  notify(data_property(S, P, DataTerm)).
```

From your comment in [#289](https://github.com/knowrob/knowrob/issues/289#issuecomment-817593125), it seems the `notify` predicate is gonna be replaced by the new `watch` module. Maybe we could get this merged in the meantime?
Thanks!

daniel86, 2021-05-14:
Yes this is possible with watch which is currently limited to watching triples.
You could watch a property, or subject/object. then react on the generated events.
Currently I think deletions are not reported yet by watch.

daniel86, 2021-05-16:
closing the ticket as the `watch` interface was merged into master

closed 2021-05-16 by daniel86