knowrob/knowrob #304: Prolog error in tf_mng_tree/2: `lookup_transforms/2` doesn't seem to be accessable

JeremiasThun, 2021-09-13:
Hi,
I am trying to run the predicate `tf_mng_tree/2` from `f_mongo` module, but prolog returns an `existence_error` for a predicate inside of `tf_mng_tree/2`:

    ?- tf_mng_tree(0.0, Tree)
    Prolog query failed: error(existence_error(procedure,:(tf_mongo,/(lookup_transforms,2))),context(:(tf_mongo,/(tf_mng_tree,2)),_370))

In line 229 of `tf_mongo.pl`, the predicate `lookup_transforms/2` is run and doesn't seem to be accessable:
https://github.com/knowrob/knowrob/blob/4d7af5484683cf2ff0aedd3b0126ae8c5077e035/src/ros/tf/tf_mongo.pl#L227-L238

benjaminalt, 2021-09-28:
I came across the same thing recently and created a PR with a fix for this: https://github.com/knowrob/knowrob/pull/307

sasjonge, 2022-07-26:
This was fixed by https://github.com/knowrob/knowrob/pull/307

closed 2022-07-26 by sasjonge