knowrob/knowrob #305: How to access mongo without access control?

JeremiasThun, 2021-09-27:
The KnowRob MongoInterface knows two ways to communicate with mongo: either via localhost with no specifications at all or with authentication and everything. But what if I want to connect to a MongoDB with a host other than localhost and no access control (as is the default for mongo installations)? 

It doesn't seem to be possible right now.

https://github.com/knowrob/knowrob/blob/4d7af5484683cf2ff0aedd3b0126ae8c5077e035/src/db/mongo/MongoInterface.cpp#L97-L109

JeremiasThun, 2021-09-27:
In my case, this other mongo host is a docker container, running in a GitLab CI/CD pipeline as Service.

daniel86, 2024-04-02:
The handling is better now, user and password are optional, see https://github.com/knowrob/knowrob/blob/dev/src/storage/mongo/MongoKnowledgeGraph.cpp#L173

closed 2024-04-02 by daniel86