knowrob/knowrob #309: Error on launching " roslaunch knowrob knowrob.launch"

divishad12, 2021-10-01:
On launching the knowrob launch file, I am getting following error.

SUMMARY
========

PARAMETERS
* /initial_goal: true
* /initial_package: knowrob
* /num_pl_threads: 2
* /num_ros_threads: 4
* /rosdistro: noetic
* /rosversion: 1.15.11

NODES
/
  json_wrapper (rosprolog/json_ros.py)
  rosprolog (rosprolog/rosprolog_node)

ROS_MASTER_URI=http://localhost:11311

process[rosprolog-1]: started with pid [144267]
process[json_wrapper-2]: started with pid [144268]
[ INFO] [1633099988.181769136]: The knowledge base is starting up, that may take a few moments!
[ INFO] [1633099988.188465456]: mng_db_name(roslog)
[INFO] [1633099988.204051]: json_wrapper service is running

**Warning: /home/divishad/ros_ws/src/knowrob/src/__init__.pl:52:
Warning:    Goal (directive) failed: user:knowrob:initialize_db**

benjaminalt, 2021-10-06:
Maybe this is related to https://github.com/knowrob/knowrob/issues/310? You could try the workaround there of manually building and installing swi-prolog 8.2.4.

sasjonge, 2022-07-22:
@divishad12 Do you still have this problem, or could we close this issuue?

closed 2022-07-26 by sasjonge