knowrob/knowrob #320: Troubles following tutorials

jpcoffelt, 2021-12-20:
I am trying to follow the tutorial [here ](http://www.knowrob.org/doc/knowrob_basics), but on the step:

`?- tripledb_load('package://iai_semantic_maps/owl/kitchen.owl').`

I receive the error message:

`ERROR: Unknown procedure: tripledb_load/1 (DWIM could not correct goal)`.

When attempting to troubleshoot the problem, I was also unable to locate the _knowrob_map_data_ package mentioned in the preceding sentence.

Any help is greatly appreciated!

jpcoffelt, 2021-12-20:
As a followup, any chance the deprecated tutorials [here ](http://www.knowrob.org/doc) might be updated, especially "Reasoning about objects" and "Create a model of your robot in the Semantic Robot Description Language (SRDL)"? They seem like they were well made and would be very useful to current KnowRob newbs.

mahyamkashani, 2022-01-11:
> `ERROR: Unknown procedure: tripledb_load/1 (DWIM could not correct goal)`.

 Have you loaded the code into the interpreter? ['/file/path'].
you can write prolog code into pl file and the load them in command line.

jpcoffelt, 2022-01-12:
I think the issue is that the command/macro/function/whatever "tripledb_load" is not recognized. The OWL file can be located, but the command "tripledb_load" appears to be deprecated.

What code should I put into the .pl file?

mahyamkashani, 2022-01-16:
> I think the issue is that the command/macro/function/whatever "tripledb_load" is not recognized. The OWL file can be located, but the command "tripledb_load" appears to be deprecated.
> 
> What code should I put into the .pl file?

I got it wrong. tripledb_load can not work for me also when I want to query on it.
This predicate should use to load ontology:
load_owl('owl file inyour catkinized package', namespace[give a name]).
like:
load_owl('package://knowrob/owl/maps/iai_room_v1',[namespace(room)]).

this works for me.

nitin5, 2022-05-04:
Hi 
I am unable to run the example as well. I tried command posted by Mahyamkashani as well but no luck. Any places where updated tutorials are available?

```
[debug]  ?- load_owl('package://knowrob/owl/maps/iai_room_v1',[namespace(room1)]).
ERROR: source_sink `/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1' does not exist (No such file or directory)
ERROR: In:
ERROR:   [23] open(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,read,_10170,[type(binary)])
ERROR:   [22] iostream:open_any_builtin(file(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1),read,_10254,close(_10268),[dialect(xmlns),...|...],[dialect(xmlns),...|...]) at /usr/lib/swi-prolog/library/iostream.pl:203
ERROR:   [21] iostream:open_any(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,read,_10398,_10400,[dialect(xmlns),...|...]) at /usr/lib/swi-prolog/library/iostream.pl:154
ERROR:   [20] <meta call>
ERROR:   [19] $sig_atomic(sgml:open_any(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,read,_10650,_10652,...)) <foreign>
ERROR:   [18] setup_call_catcher_cleanup(sgml:open_any(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,read,_10854,_10856,...),sgml:load_structure_from_stream(_10830,...,...),_10818,sgml:close_any(_10808)) at /usr/lib/swi-prolog/boot/init.pl:647
ERROR:   [17] setup_call_cleanup(sgml:open_any(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,read,_10986,_10988,...),sgml:load_structure_from_stream(_10962,...,...),sgml:close_any(_10946)) at /usr/lib/swi-prolog/boot/init.pl:651
ERROR:   [16] sgml:load_structure(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,[_11086],rdf:[...|...]) at /usr/lib/swi-prolog/library/sgml.pl:393
ERROR:   [15] rdf:load_rdf(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,_11224,lang_db:[...]) at /usr/lib/swi-prolog/library/rdf.pl:104
ERROR:   [14] lang_db:load_rdf_1(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,_11368) at /home/cair/code_apr22/src/knowrob/knowrob/src/lang/db.pl:383
ERROR:   [13] lang_db:load_rdf_(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,_11512) at /home/cair/code_apr22/src/knowrob/knowrob/src/lang/db.pl:378
ERROR:   [12] lang_db:load_owl0(/home/cair/code_apr22/src/knowrob/knowrob/owl/maps/iai_room_v1,dict{time:dict{since: ...,until: ...}},iai_room_v1,common) at /home/cair/code_apr22/src/knowrob/knowrob/src/lang/db.pl:252
ERROR:   [11] lang_db:load_owl(package://knowrob/owl/maps/iai_room_v1,dict{time:dict{since: ...,until: ...}},common) at /home/cair/code_apr22/src/knowrob/knowrob/src/lang/db.pl:220
ERROR:   [10] lang_db:load_owl(package://knowrob/owl/maps/iai_room_v1,[namespace(room1)]) at /home/cair/code_apr22/src/knowrob/knowrob/src/lang/db.pl:195
ERROR:    [9] toplevel_call(user:user: ...) at /usr/lib/swi-prolog/boot/toplevel.pl:1117

```

nitin5, 2022-05-04:
iai_room_v1.owl file exists but i still get the same error:

```
|     load_owl('package://knowrob/owl/maps/iai_room_v1.owl',[namespace(room1)]).
ERROR: Unknown procedure: load_owl/2 (DWIM could not correct goal)
[debug]  ?- 

```

mahyamkashani, 2022-05-04:
> iai_room_v1.owl file exists but i still get the same error:
> 
> ```
> |     load_owl('package://knowrob/owl/maps/iai_room_v1.owl',[namespace(room1)]).
> ERROR: Unknown procedure: load_owl/2 (DWIM could not correct goal)
> [debug]  ?- 
> ```

Hi, 
Which branch are you using? Master or other branches?

Please first check that MongoDB is running or not by using:

`sudo systemctl status mongodb.service`

if it is not active, please restart it by using:

`sudo systemctl start mongodb.service`

make sure that you already call below code in each terminal:

`source ~/catkin_ws (your catkin workspace that you create ros packages)/devel/setup.bash `

then launch knowrob:

`roslaunch knowrob knowrob.launch`

and then launching rosprolog commandline:

`rosrun rosprolog rosprolog_commandline.py`

you can see that what kind of namespaces are already exist like as follows:

> dc: http://purl.org/dc/elements/1.1/
> dcterms: http://purl.org/dc/terms/
> eor: http://dublincore.org/2000/03/13/eor#
> foaf: http://xmlns.com/foaf/0.1/
> owl: http://www.w3.org/2002/07/owl#
> rdf: http://www.w3.org/1999/02/22-rdf-syntax-ns#
> rdfs: http://www.w3.org/2000/01/rdf-schema#
> serql: http://www.openrdf.org/schema/serql#
> skos: http://www.w3.org/2004/02/skos/core#
> void: http://rdfs.org/ns/void#
> xsd: http://www.w3.org/2001/XMLSchema#
> qudt: http://data.nasa.gov/qudt/owl/qudt#
> dul: http://www.ontologydesignpatterns.org/ont/dul/DUL.owl#
> io: http://www.ontologydesignpatterns.org/ont/dul/IOLite.owl#
> soma: http://www.ease-crc.org/ont/SOMA.owl#
> knowrob: http://knowrob.org/kb/knowrob.owl#
> test: http://knowrob.org/kb/swrl_test#
> urdf: http://knowrob.org/kb/urdf.owl#

or you can use this predicate to see which rdf files are already loaded:

`rdf_current_ns(Loaded,_).`

 
finally, try this one:

`load_owl('package://knowrob/owl/maps/iai_room_v1.owl',[namespace(room1)]).`

it should be returned you True by only installing knowrob.

PS. please make sure that you used: catkin_make if you would like to build knowrob.

nitin5, 2022-05-05:
Thanks a lot. It worked for me.  
I request you to kindly guide me where I can look for updated tutorials/guidelines. I have been wandering across many pages (including depracated docs of Knowrob hoping to get something). Since I am new to KnowRob, I found OpenEase's NEEM demos difficult to setup and appreciate.

> > iai_room_v1.owl file exists but i still get the same error:
> > ```
> > |     load_owl('package://knowrob/owl/maps/iai_room_v1.owl',[namespace(room1)]).
> > ERROR: Unknown procedure: load_owl/2 (DWIM could not correct goal)
> > [debug]  ?- 
> > ```
> 
> Hi, Which branch are you using? Master or other branches?
> 
I have downloaded master branch. I have compiled it using catkin. I am using ROS Melodic.

> Please first check that MongoDB is running or not by using:
> 
> `sudo systemctl status mongodb.service`
> 

> if it is not active, please restart it by using:
> 
> `sudo systemctl start mongodb.service`
> 
I did started service using the following command  'sudo systemctl start mongod'

> make sure that you already call below code in each terminal:
> 
> `source ~/catkin_ws (your catkin workspace that you create ros packages)/devel/setup.bash `
> 
> then launch knowrob:
> 
> `roslaunch knowrob knowrob.launch`
> 
I still gets some of the errors in the console:

**auto-starting new master
process[master]: started with pid [18374]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6e87f4de-cc21-11ec-a1ab-5065f341536b
process[rosout-1]: started with pid [18424]
started core service [/rosout]
process[rosprolog-2]: started with pid [18431]
process[json_wrapper-3]: started with pid [18432]
[INFO] [1651720452.196700]: json_wrapper service is running
ERROR: /home/cair/code_apr22/src/knowrob/knowrob/src/utility/atoms.pl:16:
ERROR:    source_sink `library(delay)' does not exist
Warning: /home/cair/code_apr22/src/knowrob/knowrob/src/utility/atoms.pl:16:
Warning:    Goal (directive) failed: utils_atoms:use_module(library(delay))
[ INFO] [1651720452.354090658]: The knowledge base is starting up, that may take a few moments!
[ INFO] [1651720452.365926432]: mng_db_name(roslog)
% Parsed "unit.owl" in 0.21 sec; 7,134 triples
[ INFO] [1651720452.988663548]: detected "owl" ontology version 1648613629.0.
[ INFO] [1651720453.014790218]: detected "DUL" ontology version 1648628316.0.
[ INFO] [1651720453.016370792]: detected "IOLite" ontology version 1648628317.0.
[ INFO] [1651720453.112634346]: detected "SOMA" ontology version 1648628319.0.
[ INFO] [1651720453.323525739]: detected "knowrob" ontology version 1648613629.0.
[ INFO] [1651720453.437011878]: detected "URDF" ontology version 1648613629.0.
Warning: /home/cair/code_apr22/src/knowrob/knowrob/src/ros/marker/marker.pl:101:
Warning:    Singleton variables: [Timepoint]
[ INFO] [1651720453.467816818]: The knowledge base has finished starting up!
[ INFO] [1651720453.485005559]: rosprolog service is running.**

> and then launching rosprolog commandline:
> 

I was not running the rosprolog command line.!
I was running ' rosrun rosprolog rosprolog knowrob' and i was querying on this prompt.

> `rosrun rosprolog rosprolog_commandline.py`
>
> you can see that what kind of namespaces are already exist like as follows:
Yes
> or you can use this predicate to see which rdf files are already loaded:
> 
> `rdf_current_ns(Loaded,_).`
> 
> finally, try this one:
> 
> `load_owl('package://knowrob/owl/maps/iai_room_v1.owl',[namespace(room1)]).`
> 
> it should be returned you True by only installing knowrob.
> 
Yes it worked.
> PS. please make sure that you used: catkin_make if you would like to build knowrob.

Thanks once again.

daniel86, 2024-04-02:
Hi everybody. It is sometimes hard to keep up with documentation when development is active. I hope the last iteration has created a stable API that can last for some years.

Generally speaking, please refer to the README files and API documentation for most up to date information.
The usage tutorials are very likely outdated. I would like to include some Jupyter notebooks in this repository and also include them in CI - that would once an for all fix the problem of code and tutorials being out of sync. Maybe this year it will finally happen.

closed 2024-04-02 by daniel86