knowrob/knowrob #324: ontologydesignpatterns.org is not reachable

Rezenders, 2022-03-14:
ontologydesignpatterns.org is not reachable (maybe this is temporary) but with this `catkin_make` ifails to build knowrob package

```
-- Downloading ontology DUL-3.32.
^TCMake Error at knowrob/cmake/ontologies.cmake:48 (message):
  Failed to download ontology
  http://www.ontologydesignpatterns.org/ont/dul/DUL.owl: "Failure when
  receiving data from the peer"
Call Stack (most recent call first):
  knowrob/CMakeLists.txt:39 (install_ontology)
```

In the CMake file, wouldn't it be better to use the [ontologies github repo](https://github.com/ease-crc/dul) directly to avoid this type of problem?

Something like this:

```
install_ontology(
	URL		  "https://raw.githubusercontent.com/ease-crc/dul/master/owl/DUL.owl"
	VERSION   "3.32")
install_ontology(
	URL		  "https://raw.githubusercontent.com/ease-crc/dul/master/owl/IOLite.owl"
	VERSION   "1.8")
```

`catkin_make` completes with this

I can open a new PR if you think this is a good solution

Rezenders, 2022-03-14:
Although I am able to build the package with `catkin_make` I still get errors when trying to run `knowrob.launch`

```
 INFO] [1647273688.712034152]: detected "owl" ontology version 1647255618.0.
[ INFO] [1647273688.727498219]: detected "DUL" ontology version 1647271832.0.
[rospack] Error: package 'dul' not found
[librospack]: error while executing command
ERROR: /home/gus/knowrob_ws/src/knowrob/src/model/DUL.pl:77:
ERROR:    iri_scheme `package' does not exist
Warning: /home/gus/knowrob_ws/src/knowrob/src/model/DUL.pl:77:
Warning:    Goal (directive) failed: model_DUL:load_owl(http://www.ontologydesignpatterns.org/ont/dul/IOLite.owl,[namespace(io)])
[ INFO] [1647273688.935929731]: detected "DUL" ontology version 1647271832.0.
[rospack] Error: package 'dul' not found
[librospack]: error while executing command
ERROR: /home/gus/knowrob_ws/src/knowrob/src/model/SOMA.pl:90:
ERROR:    iri_scheme `package' does not exist
Warning: /home/gus/knowrob_ws/src/knowrob/src/model/SOMA.pl:90:
Warning:    Goal (directive) failed: model_SOMA:load_owl(http://www.ease-crc.org/ont/SOMA.owl,[namespace(soma)])
[ INFO] [1647273689.211246437]: detected "DUL" ontology version 1647271832.0.
[rospack] Error: package 'dul' not found
[librospack]: error while executing command
ERROR: /home/gus/knowrob_ws/src/knowrob/src/model/__init__.pl:14:
ERROR:    iri_scheme `package' does not exist
Warning: /home/gus/knowrob_ws/src/knowrob/src/model/__init__.pl:14:
Warning:    Goal (directive) failed: user:load_owl(http://knowrob.org/kb/knowrob.owl,[namespace(knowrob)])
[ INFO] [1647273689.436200952]: detected "DUL" ontology version 1647271832.0.
[rospack] Error: package 'dul' not found
[librospack]: error while executing command
ERROR: /home/gus/knowrob_ws/src/knowrob/src/ros/urdf/URDF.pl:50:
ERROR:    iri_scheme `package' does not exist
Warning: /home/gus/knowrob_ws/src/knowrob/src/ros/urdf/URDF.pl:50:
Warning:    Goal (directive) failed: ros_urdf:load_owl(http://knowrob.org/kb/URDF.owl,[namespace(urdf,http://knowrob.org/kb/urdf.owl#)])
Warning: /home/gus/knowrob_ws/src/knowrob/src/ros/marker/marker.pl:101:
Warning:    Singleton variables: [Timepoint]
[ INFO] [1647273689.579940012]: The knowledge base has finished starting up!
[ INFO] [1647273689.589875368]: rosprolog service is running.
```

sasjonge, 2022-03-16:
Hello Gustavo,
the link to DUL was temporarily down. It works for me now, again.
Maybe one could create a Issue for this case (a link is down), so that the build doesn't die, but instead a fork or local version of DUL is used.

sasjonge, 2022-06-10:
This PR was fixed by https://github.com/knowrob/knowrob/pull/327

We are now hosting DUL ourself.

closed 2022-06-10 by sasjonge