knowrob/knowrob #325: kb_projects doesn't fail if it should be

sasjonge, 2022-03-16:
If I call kb_project with something that should fail I still get an answer. For example the query:

kb_project(instance_of(A,dul:'Action')),writeln(A).

will print out
_310

or similar. The rosprolog_commandline tool shows 

A: _.

As a result.

sasjonge, 2022-03-23:
I could track the error down to 

https://github.com/knowrob/knowrob/blob/04793aa6a8520fceb459a834f142f9ae6ed2d85d/src/lang/terms/triple.pl#L152

Interestingly the all_ground predicate throws a instantiation error if I call

`kb_project(triple(A, rdf:type, soma:'Event')).`

but no instantiation error if I call

`kb_project(triple(A, rdf:type, soma:'Event')).`

daniel86, 2022-07-28:
@sasjonge hmm ok, so there is a predicate to generate a new name. but if the variable is not instantiated project should fail. that's your point, right?

"I could track the error down to" so all_ground succeeds where it shouldn't?

daniel86, 2024-04-02:
The issue is related to an interface that has not been migrated to the C++ re-implementation of KnowRob. We won't fix remaining issues in the legacy code.

closed 2024-04-02 by daniel86