knowrob/knowrob #328: Inconsistency in knowrob.owl

superlee1, 2022-06-10:
Hi,
there seems an issue in the knowrob.owl while reasoning with Pellet. Please find the following information copied from Protege:
""
_An error occurred during reasoning: Axiom:  Transitive: **hasPart**.
org.mindswap.pellet.exceptions.UnsupportedFeatureException: Axiom:  Transitive: hasPart
	at com.clarkparsia.pellet.owlapiv3.PelletVisitor.addUnsupportedAxiom(PelletVisitor.java:119) ~[na:na]
	at com.clarkparsia.pellet.owlapiv3.PelletVisitor.verify(PelletVisitor.java:160) ~[na:na]
	at com.clarkparsia.pellet.owlapiv3.PelletReasoner.refresh(PelletReasoner.java:969) ~[na:na]
	at com.clarkparsia.pellet.owlapiv3.PelletReasoner.<init>(PelletReasoner.java:345) ~[na:na]
	at com.clarkparsia.pellet.owlapiv3.PelletReasonerFactory.createReasoner(PelletReasonerFactory.java:69) ~[na:na]
	at com.clarkparsia.pellet.owlapiv3.PelletReasonerFactory.createReasoner(PelletReasonerFactory.java:33) ~[na:na]
	at org.protege.editor.owl.model.inference.ReasonerUtilities.createReasoner(ReasonerUtilities.java:20) ~[na:na]
	at org.protege.editor.owl.model.inference.OWLReasonerManagerImpl$ClassificationRunner.ensureRunningReasonerInitialized(OWLReasonerManagerImpl.java:428) ~[na:na]
	at org.protege.editor.owl.model.inference.OWLReasonerManagerImpl$ClassificationRunner.run(OWLReasonerManagerImpl.java:386) ~[na:na]
	at java.lang.Thread.run(Thread.java:745) ~[na:1.8.0_121]
""

sasjonge, 2022-06-10:
Hi,

could you tell me how you open knowrob in protege? Do you open a local version or the one at the URI of knowrob? 

Which version of pellet are you using? I couldn't reproduce the error on my protege with pellet.

superlee1, 2022-06-14:
Hi, thanks for the timely feedback! Please find the required info:

actions: download git repo -> open the knowrob.owl in the owl folder -> Protege prompted for missing ontology of "URDF" -> import local URDF.owl -> successfully opened in Protege -> run Pellet -> same inconsistency error reproduced

Protege version: 5.5
Pellet version: 2.2.0 (default)

Thanks for your effort!
:D

sasjonge, 2022-07-20:
I could reproduce the error with Pellet, but other reasoners are not throwing this error. The definitions stem from DUL and SOMA which a checked for their consistency in the CI of the SOMA repository (see for example [here](https://github.com/ease-crc/soma/runs/7409078670?check_suite_focus=true)).

I believe Pellet is not any more maintained, so the error could be within Pellet.

daniel86, 2022-11-19:
I am closing this issue due to inactivity. please re-open if there are still problems

closed 2022-11-19 by daniel86