knowrob/knowrob #329: Is URDF2SRDL deprecated?

Aqudi, 2022-06-28:
I'm trying to make a model of my robot using URDF2SRDL, but there is no file in master branch. 
Is it no longer used?

Aqudi, 2022-07-03:
Thank you for your response.
Then, is there a way to replace URDF2SRDL?

daniel86, 2022-07-05:
Hello. Have a look here: https://github.com/knowrob/knowrob/blob/master/launch/map.launch
It loads PR2 OWL file + URDF. Then you basically have kinematics from URDF plus components defined in the OWL file.
The whole SRDL ontology has not been aligned yet with the SOMA ontology used by KnowRob since some time, but it is high priority to work on capability modelling.

Aqudi, 2022-07-07:
> Hello. Have a look here: https://github.com/knowrob/knowrob/blob/master/launch/map.launch It loads PR2 OWL file + URDF. Then you basically have kinematics from URDF plus components defined in the [...]

Thank you!

closed 2022-07-07 by Aqudi