knowrob/knowrob #330: Error on startup - iri_scheme `http' does not exist

Janfiderheld, 2022-06-28:
Hello,
I followed the tutorial for the installation without bigger problems. Now I'm trying to start KnowRob using 
`roslaunch knowrob knowrob.launch`

However, the following output is returned:
```
started roslaunch server http://LaptopJanPhilipp:43507/

SUMMARY
========

PARAMETERS
 * /initial_goal: urdf_init
 * /initial_package: knowrob
 * /mongodb_uri: mongodb://localho...
 * /num_pl_threads: 2
 * /num_ros_threads: 4
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    json_wrapper (rosprolog/json_ros.py)
    rosprolog (rosprolog/rosprolog_node)

ROS_MASTER_URI=http://localhost:11311

process[rosprolog-1]: started with pid [19667]
process[json_wrapper-2]: started with pid [19668]
[ INFO] [1656413224.263933591]: The knowledge base is starting up, that may take a few moments!
[ INFO] [1656413224.271236144]: mng_db_name(roslog)
[INFO] [1656413224.284122]: json_wrapper service is running
% Parsed "unit.owl" in 0.12 sec; 7,134 triples
[ INFO] [1656413224.609418682]: detected "owl" ontology version 1655968594.0.
[ INFO] [1656413224.624520073]: detected "DUL" ontology version 1655969531.0.
[ INFO] [1656413224.722827410]: detected "DUL" ontology version 1655969531.0.
[ INFO] [1656413225.265498981]: detected "DUL" ontology version 1655969531.0.
ERROR: /home/user/catkin_ws/src/knowrob/src/model/SOMA.pl:90:
ERROR:    iri_scheme `http' does not exist
Warning: /home/user/catkin_ws/src/knowrob/src/model/SOMA.pl:90:
Warning:    Goal (directive) failed: model_SOMA:load_owl(http://www.ease-crc.org/ont/SOMA.owl,[namespace(soma)])
[ INFO] [1656413225.580579617]: detected "DUL" ontology version 1655969531.0.
[ INFO] [1656413226.052440765]: detected "DUL" ontology version 1655969531.0.
ERROR: /home/user/catkin_ws/src/knowrob/src/model/__init__.pl:14:
ERROR:    iri_scheme `http' does not exist
Warning: /home/user/catkin_ws/src/knowrob/src/model/__init__.pl:14:
Warning:    Goal (directive) failed: user:load_owl(http://knowrob.org/kb/knowrob.owl,[namespace(knowrob)])
[ INFO] [1656413226.293482015]: detected "DUL" ontology version 1655969531.0.
[ INFO] [1656413226.750998124]: detected "DUL" ontology version 1655969531.0.
ERROR: /home/user/catkin_ws/src/knowrob/src/ros/urdf/URDF.pl:50:
ERROR:    iri_scheme `http' does not exist
Warning: /home/user/catkin_ws/src/knowrob/src/ros/urdf/URDF.pl:50:
Warning:    Goal (directive) failed: ros_urdf:load_owl(http://knowrob.org/kb/URDF.owl,[namespace(urdf,http://knowrob.org/kb/urdf.owl#)])
Warning: /home/user/catkin_ws/src/knowrob/src/ros/marker/marker.pl:101:
Warning:    Singleton variables: [Timepoint]
[ INFO] [1656413226.901038971]: The knowledge base has finished starting up!
[ INFO] [1656413226.918007073]: rosprolog service is running.
```

Afterwards, the process is frozen and I cannot interact with it except for killing it. Am I missing something?
Here are some more details about my setup:

- Ubuntu 20.04.4 LTS 
- ROS Noetic v1.15.14
- SWI Prolog v8.4.3
- MongoDB v4.4.15
- I've used the 'noetic' branch for KnowRob and rosprolog

Thank you for your help!

Janfiderheld, 2022-06-30:
Update:
I tried to manually register 'http' as an iri scheme by adding the following lines to the [db.pl](https://github.com/knowrob/knowrob/blob/master/src/lang/db.pl):
``` Prolog
:- register_iri_scheme('http', iri_hook, []).

iri_hook(open(Mode, Options), IRI, Stream) :-
    open(IRI, Mode, Stream, Options).
```

With this change, KnowRob behaves the same way during startup but the error message changed from
`ERROR:    iri_scheme 'http' does not exist`
to
`ERROR:    lang_db:version_matcher/2: Unknown procedure: lang_db:digits/3`
for all 3 occurrences in my original comment. 

Am I on the right track? Is this a sensible way to solve my problem going forward? Do you have any ideas how to handle this new error message?
I would appreciate any help. Thanks!

daniel86, 2022-07-05:
Hello. sorry for the delay. Could it be that you need to update your Prolog version? 
SWI Prolog >= 8.2.4 is written in the readme

sasjonge, 2022-07-06:
Hello,

we merged a PR that could have potentially fixed this issue:

https://github.com/knowrob/knowrob/pull/332

Could you test if the error still occurs for you?

Janfiderheld, 2022-07-06:
Hey, that change fixed my problem and the errors are gone. Thank you very much! :)

However, I still have the problem that the process freezes after
`[ INFO] [1657119056.771807552]: rosprolog service is running.`
Is this another occurrence of the bug described in #316?

sasjonge, 2022-07-06:
This is the last message you get from the roslaunch. Can you send queries over the rosprolog_commandline? You start the commandline with this command:

`rosrun rosprolog rosprolog_commandline.py`

Janfiderheld, 2022-07-06:
Ah, now I understand. I thought you were supposed to start the commandline afterwards in the same terminal window. When I'm using a different terminal, everything works fine and I can send queries. Thanks for your help!

closed 2022-07-06 by Janfiderheld