knowrob/knowrob #338: Error when launching **roslaunch knowrob knowrob.launch**

MinchoU, 2022-07-25:
My development is Ubuntu 20.04 and ROS noetic. I'm installing noetic version of knowrob from fork of https://github.com/hawkina/knowrob. But I'm facing error when launching command 

**roslaunch knowrob knowrob.launch**

Error message is shown like this : 

Warning: /home/minu/perception_ws/src/knowrob/src/__init__.pl:52:
Warning:    Goal (directive) failed: user:knowrob:initialize_db
ERROR: /home/minu/perception_ws/src/knowrob/src/lang/db.pl:149:
ERROR:    Unhandled exception: mng_error(collection_remove('No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on \'localhost:27017\']'))
[ERROR] [1658747314.803292608]: mng_error(collection_remove(No suitable servers found: `serverSelectionTimeoutMS` expired: [connection refused calling ismaster on 'localhost:27017']))
[ERROR] [1658747314.803383592]: Failed to load initial_package knowrob.
[ERROR] [1658747314.803416192]: rosprolog service failed to initialize.
[rosprolog-2] process has died [pid 168085, exit code 1, cmd /home/minu/perception_ws/devel/lib/rosprolog/rosprolog_node __name:=rosprolog __log:=/home/minu/.ros/log/ea25b770-0c09-11ed-b71d-53ba4621ef86/rosprolog-2.log].

Can I get some help? Thanks :)

sasjonge, 2022-07-25:
Hey, this sounds like the MongoDB Server is not running. Did you install MongoDB (see [here](https://github.com/knowrob/knowrob#installation-of-swi-prolog-and-mongodb)? If yes, then mongo is potentially not working. You could start it with `sudo systemctl start mongod.service`

MinchoU, 2022-07-26:
Thanks! It worked perfectly.

closed 2022-07-26 by MinchoU