knowrob/knowrob #339: How to save perception data using computable

Aqudi, 2022-07-28:
Hello, I want to store recognition results such as object detection and face detection using Computable. 
However, I cannot find any documents on this. How do I use this feature?

And if I use computable predicate, would the perception data be stored in mongodb too?

daniel86, 2022-07-28:
Hello @Aqudi, what form do the recognition results have? what is the information content?
Generally speaking, the results can be conceptualized in terms of the ontology in which case the data would end up in the triple storage which is currently implemented with MongoDB. If you rather want to quantify the results, than you could encode the data in JSON and also load into MongoDB. In this case you can use some general predicates in KnowRob for querying the database, or define a set of specialized predicates that are implemented using the database predicates KnowRob provides.

Aqudi, 2022-08-10:
Hello @daniel86, thank you for your response!

I'm currently studying SOMA, DUL, and Knowrob to convert data to ontology format. 

I haven't decided exactly what data to use yet. However, I believe that sensor information such as object detection and battery condition will be included, so I will leave a rough shape below.

**object detection**
```
{
   state: "READY/ERROR/ENABLED",
   objectType: {
        depth,
        point: { x1, x2, y1, y2 }.
        value: "Human/Car/..."
   }
}
```

**battery status**
```
{
    state: AC/CARGHING/BATTERY,
    chargeRate:0.87
}
```

Aqudi, 2022-08-10:
I have additional questions about the process of utilizing Knowrob.

I want to input sensor data in the same way as the picture below. Is this process correct?

[image]

And below is a example of computable predicate `comp_nearby` that calculates whether two objects are close or not. Is it right to use it?

[image]

daniel86, 2022-10-12:
@Aqudi yes, in principle. Saving results of the computable to MongoDB might be a bit problematic at the moment.
Also regarding the positions, I usually use TF message format of ROS without storing the position data in the knowledge graph. Only qualitative spatial relations I store in the graph. There are some predicated in KnowRob to interact with TF, and a small implementation of a belief state where the poses of objects are stores and published on the TF topic.
To log TF messages, I use the ROS mongo DB logging tool: http://wiki.ros.org/mongodb_store
KnowRob can work with TF data stored in mongo using the mentioned tool

nitin5, 2022-11-28:
Greetings,
@Aqudi @daniel86 
I am also trying to do the same i.e. take data from simulator, run perception modules and perform some planning activity. I am looking replicate the KnowRob2 paper's simulation. 
Any pointers in this direction will be great help. I am pretty new to Prolog but have background in other programming languages. 

Thank you.

daniel86, 2024-04-02:
@nitin5 sorry for late response, last months were very busy as I did a re-write of KnowRob over the last year and now I am kind of done with it and try to clean up a bit.

Regarding simulation: this is not really something considered within this repository. There are a couple of people working on the simulation part outlined in knowrob2 paper who are part of Michael Beetz group in Bremen. maybe these are two references you can check:

Michael Neumann, Andrei Haidu and Michael Beetz, ["URoboSim — a simulation-based predictive modelling engine forcognition-enabled robot manipulation"](https://embodied-ai.org/papers/URoboSim.pdf), In 2nd Embodied AI Workshop (CVPR), 2021. [[bibtex]](https://ai.uni-bremen.de/team/bibtexbrowser.php?key=urobosim21cvpr&bib=bibfiles%2Fallpublications.bib) [[pdf]](https://embodied-ai.org/papers/URoboSim.pdf)

Patrick Mania, Franklin Kenghagho Kenfack, Michael Neumann and Michael Beetz, "Imagination-enabled Robot Perception", In International Conference on Intelligent Robots and Systems (IROS), 2021. Best Paper Award on Cognitive Robotics [[bibtex]](https://ai.uni-bremen.de/team/bibtexbrowser.php?key=mania21imaginationenabled&bib=bibfiles%2Fallpublications.bib)

closed 2024-04-02 by daniel86