knowrob/knowrob #358: Installation error: catkin_make fails  (can't download SOMA-HOME-current)

Dorteel, 2022-11-23:
Hi,

I'm trying to install KnowRob following the instructions, but getting a timeout when running catkin_make. 
```console
-- Downloading ontology SOMA-current.
-- Downloading ontology SOMA-HOME-current.
CMake Error at knowrob/cmake/ontologies.cmake:48 (message):
  Failed to download ontology
  https://ease-crc.github.io/soma/owl/current/SOMA-HOME.owl: "Timeout was
  reached"
Call Stack (most recent call first):
  knowrob/CMakeLists.txt:45 (install_ontology)

-- Configuring incomplete, errors occurred!
See also "/home/dorteel/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dorteel/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```
Do you have any suggestions? I'm not sure why this happens, as I can access [SOMA-HOME](https://ease-crc.github.io/soma/owl/current/SOMA-HOME.owl) through my browser.
Thanks in advance!

nitin5, 2022-11-28:
I am able to install Knowrob today. 

Are you able to open link 'https://ease-crc.github.io/soma/owl/current/SOMA-HOME.owl' in browser? Just to check whether you have connectivity to the site.

daniel86, 2024-04-02:
Probably was a temporary outage of some server.

closed 2024-04-02 by daniel86