knowrob/knowrob #390: Built-in properties like subClassOf not working on terminal

sasjonge, 2024-04-22:
Hey,

I tried to run a subClassOf query (with the default.json as settings file) on the terminal, but it failed. I found part of the error and fixed it for the MongoBackend (PR follows after I created this issue), but I have still problems with the RedlandModel as a backend.

### Issue

The query was:

`triple(swrl_test:'Adult', rdfs:'subClassOf', swrl_test:'TestThing')
`

This will result in the following output:

`[15:14:29.325] [warning] Predicate rdfs:subClassOf(test:Adult,test:TestThing) is neither materialized in EDB nor defined by a reasoner.
don't know`

The same query runs through in the unit tests, [here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/tests/StorageTest.cpp#L194).

I also tested it with the RedlandModel as backend, with the same result.

### Cause

The function isMaterializedInEDB returns false [here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/queries/QueryPipeline.cpp#L217). This is because the vocabulary frequency is 0 for rdfs:subclassOf (see [here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/queries/QueryPipeline.cpp#L41)).

There are at least two causes at play, which I already could fix:

1. [Here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/KnowledgeBase.cpp#L172) only persistent backends are looked at. If one uses RedlandModel with e.g. Memory as backend, it is not seen as a persistent backend. I would propose to iterate over all queryable backends instead of only persistent ones.
2. The frequency is not increased for properties like rdfs:subClassOf or rdf:type. I fixed it by adding a increaseFrequencyCall [here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/KnowledgeBase.cpp#L186).

This fixed the problem for the MongoBackend (and Unit Tests are still running), but for the RedlandModel the query will after the two fixes even result in a crash. This seems to be because RedlandModel has a problem with e.g. the rdfs:subClassOf(?x,?y) query, generate [here](https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/KnowledgeBase.cpp#L189).

sasjonge, 2024-04-22:
I created a PR that contains atleast the fix that works with the mongo backend.

daniel86, 2024-04-22:
About 1. This is not a problem because non-persistent storages are supposed to be empty initially - so nothing to loop over. So it is perfectly fine to loop only over the persistent ones. If you load an OWL file during runtime, the vocabulary should be updated accordingly.

daniel86, 2024-04-22:
The runtime update of the Vocabulary is done by the `Transaction` class here: https://github.com/knowrob/knowrob/blob/e94c846d8d6734eebc48db5b0055c31d2df42a3c/src/storage/Transaction.cpp#L187

sasjonge, 2024-04-22:
Ahh, then it seems that I should add an increaseFrequency in updateVocabulary for e.g. subclassOf? 

I will update the PR accordingly.

daniel86, 2024-04-22:
yes, there is a case already for subclassOf where you can add a call. directly the first one

sasjonge, 2024-04-22:
I updated the [PR](https://github.com/knowrob/knowrob/pull/391)  and it now works with all backends.

daniel86, 2024-04-22:
merged

closed 2024-04-22 by daniel86