knowrob/knowrob #394: Memory Issue with mongo storage engine

McModknower, 2024-05-14:
When using the mongo storage engine, parts of the memory might be freed before using them.

I got this to cause a crash by asking knowrob-terminal to prove the implication `rdf:type(A,rdfs:Class) -> rdf:type(B,A)` directly after the start, however any sufficiently complex formula should be enough to cause inconsistent behaviour.

Using valgrind as `valgrind --log-file=valgrind.log --num-callers=24 --track-origins=yes ./knowrob-terminal --config ../settings/default.json` and entering the query above followed by exit i got the attached logfile.

[valgrind.log](https://github.com/knowrob/knowrob/files/15309799/valgrind.log)

The following part indicates to me that the problem is that each mongo response part allocates memory, knowrob points to data from there and sends these pointers to another thread, and then the thread for mongo responses frees the memory because it processes the next part of the response.
The other thread uses the pointers, even though the mongo thread might have already freed the memory they point to.

```
==3993== Invalid read of size 1
==3993==    at 0x612BCFD: std::char_traits<char>::eq(char const&, char const&) (char_traits.h:304)
==3993==    by 0x612BEFB: std::basic_string_view<char, std::char_traits<char> >::rfind(char, unsigned long) const (string_view.tcc:113)
==3993==    by 0x63400DF: knowrob::semweb::Resource::iri_ns(std::basic_string_view<char, std::char_traits<char> >, bool) (Resource.cpp:65)
==3993==    by 0x6335070: knowrob::IRIAtom::write(std::ostream&) const (IRIAtom.cpp:42)
==3993==    by 0x60027E5: knowrob::TermWriter::TermWriter(knowrob::Term const&, std::ostream&) (Term.h:127)
==3993==    by 0x600206B: std::operator<<(std::ostream&, knowrob::Term const&) (Term.cpp:71)
==3993==    by 0x6055CDD: knowrob::Predicate::write(std::ostream&) const (Predicate.cpp:68)
==3993==    by 0x60560B8: std::operator<<(std::ostream&, knowrob::Predicate const&) (Predicate.cpp:108)
==3993==    by 0x6288AC1: knowrob::AnswerYes::stringFormOfYes[abi:cxx11]() const (AnswerYes.cpp:114)
==3993==    by 0x6084CC4: knowrob::Answer::stringFormOfAnswer[abi:cxx11]() const (Answer.cpp:95)
==3993==    by 0x62A3E8E: knowrob::Token::stringForm[abi:cxx11]() const (Token.cpp:35)
==3993==    by 0x62A3F53: std::operator<<(std::ostream&, knowrob::Token const&) (Token.cpp:41)
==3993==    by 0x1C2BB4: KnowRobTerminal::pushQueryResult(std::shared_ptr<knowrob::Answer const> const&) (terminal.cpp:203)
==3993==    by 0x1C37AA: KnowRobTerminal::runQuery(std::shared_ptr<knowrob::FormulaQuery const> const&) (terminal.cpp:287)
==3993==    by 0x1C32A3: KnowRobTerminal::runQuery(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (terminal.cpp:254)
==3993==    by 0x1C3FBC: KnowRobTerminal::enter() (terminal.cpp:341)
==3993==    by 0x1C5F3C: KnowRobTerminal::run() (terminal.cpp:618)
==3993==    by 0x1BE2F7: run(int, char**) (terminal.cpp:685)
==3993==    by 0x1BE5C2: main (terminal.cpp:693)
==3993==  Address 0xfa65a7f is 191 bytes inside a block of size 32,768 free'd
==3993==    at 0x483CA3F: free (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==3993==    by 0x713412B: bson_destroy (in /usr/lib/x86_64-linux-gnu/libbson-1.0.so.0.0.0)
==3993==    by 0x70BF8D1: _mongoc_cursor_response_refresh (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BFC7B: ??? (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BDF1C: mongoc_cursor_next (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x6254E56: knowrob::mongo::Cursor::next(_bson_t const**, bool) (Cursor.cpp:79)
==3993==    by 0x6270E3E: knowrob::mongo::BindingsCursor::nextBindings(std::shared_ptr<knowrob::Bindings> const&) (BindingsCursor.cpp:71)
==3993==    by 0x625B18C: knowrob::MongoKnowledgeGraph::query(std::shared_ptr<knowrob::GraphQuery> const&, std::function<void (std::shared_ptr<knowrob::Bindings const> const&)> const&) (MongoKnowledgeGraph.cpp:400)
==3993==    by 0x63A7B88: knowrob::StorageInterface::query(std::shared_ptr<knowrob::QueryableStorage> const&, std::shared_ptr<knowrob::GraphQuery> const&, std::function<void (std::shared_ptr<knowrob::Bindings const> const&)> const&) const (StorageInterface.cpp:335)
==3993==    by 0x63A7E29: knowrob::StorageInterface::pushIntoCursor(std::shared_ptr<knowrob::QueryableStorage> const&, std::shared_ptr<knowrob::GraphPathQuery> const&, std::shared_ptr<knowrob::TokenBuffer> const&) const (StorageInterface.cpp:345)
==3993==    by 0x63A80DD: knowrob::StorageInterface::getAnswerCursor(std::shared_ptr<knowrob::QueryableStorage> const&, std::shared_ptr<knowrob::GraphPathQuery> const&)::{lambda(std::function<bool ()> const&)#1}::operator()(std::function<bool ()> const&) const (StorageInterface.cpp:366)
==3993==    by 0x63AAF17: std::_Function_handler<void (std::function<bool ()> const&), knowrob::StorageInterface::getAnswerCursor(std::shared_ptr<knowrob::QueryableStorage> const&, std::shared_ptr<knowrob::GraphPathQuery> const&)::{lambda(std::function<bool ()> const&)#1}>::_M_invoke(std::_Any_data const&, std::function<bool ()> const&) (std_function.h:300)
==3993==    by 0x61A6976: std::function<void (std::function<bool ()> const&)>::operator()(std::function<bool ()> const&) const (std_function.h:688)
==3993==    by 0x61A645C: knowrob::ThreadPool::LambdaRunner::run() (ThreadPool.h:155)
==3993==    by 0x5F908C8: knowrob::ThreadPool::Runner::runInternal() (ThreadPool.cpp:169)
==3993==    by 0x5F90522: knowrob::ThreadPool::Worker::run() (ThreadPool.cpp:129)
==3993==    by 0x5F96AAE: void std::__invoke_impl<void, void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*>(std::__invoke_memfun_deref, void (knowrob::ThreadPool::Worker::*&&)(), knowrob::ThreadPool::Worker*&&) (invoke.h:73)
==3993==    by 0x5F969A0: std::__invoke_result<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*>::type std::__invoke<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*>(void (knowrob::ThreadPool::Worker::*&&)(), knowrob::ThreadPool::Worker*&&) (invoke.h:95)
==3993==    by 0x5F968C2: void std::thread::_Invoker<std::tuple<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) (thread:244)
==3993==    by 0x5F96769: std::thread::_Invoker<std::tuple<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*> >::operator()() (thread:251)
==3993==    by 0x5F96697: std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*> > >::_M_run() (thread:195)
==3993==    by 0x69F6DF3: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==3993==    by 0x6905608: start_thread (pthread_create.c:477)
==3993==    by 0x6C3C352: clone (clone.S:95)
==3993==  Block was alloc'd at
==3993==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==3993==    by 0x713F592: bson_malloc (in /usr/lib/x86_64-linux-gnu/libbson-1.0.so.0.0.0)
==3993==    by 0x7133FFA: bson_copy_to (in /usr/lib/x86_64-linux-gnu/libbson-1.0.so.0.0.0)
==3993==    by 0x70B070A: ??? (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70B1778: mongoc_cluster_run_command_monitored (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BD8B1: _mongoc_cursor_run_command (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BF8E5: _mongoc_cursor_response_refresh (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BFD5F: ??? (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x70BDF1C: mongoc_cursor_next (in /usr/lib/x86_64-linux-gnu/libmongoc-1.0.so.0.0.0)
==3993==    by 0x6254E56: knowrob::mongo::Cursor::next(_bson_t const**, bool) (Cursor.cpp:79)
==3993==    by 0x6270E3E: knowrob::mongo::BindingsCursor::nextBindings(std::shared_ptr<knowrob::Bindings> const&) (BindingsCursor.cpp:71)
==3993==    by 0x625B18C: knowrob::MongoKnowledgeGraph::query(std::shared_ptr<knowrob::GraphQuery> const&, std::function<void (std::shared_ptr<knowrob::Bindings const> const&)> const&) (MongoKnowledgeGraph.cpp:400)
==3993==    by 0x63A7B88: knowrob::StorageInterface::query(std::shared_ptr<knowrob::QueryableStorage> const&, std::shared_ptr<knowrob::GraphQuery> const&, std::function<void (std::shared_ptr<knowrob::Bindings const> const&)> const&) const (StorageInterface.cpp:335)
[... cut ...]
==3993==    by 0x5F969A0: std::__invoke_result<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*>::type std::__invoke<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*>(void (knowrob::ThreadPool::Worker::*&&)(), knowrob::ThreadPool::Worker*&&) (invoke.h:95)
==3993==    by 0x5F968C2: void std::thread::_Invoker<std::tuple<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) (thread:244)
==3993==    by 0x5F96769: std::thread::_Invoker<std::tuple<void (knowrob::ThreadPool::Worker::*)(), knowrob::ThreadPool::Worker*> >::operator()() (thread:251)
```

McModknower, 2024-05-15:
`knowrob::mongo::Cursor::next(_bson_t const**, bool) (Cursor.cpp:79)` calls `mongoc_cursor_next`. Its documentation at https://mongoc.org/libmongoc/current/mongoc_cursor_next.html#lifecycle states:
> The bson objects set in this function are ephemeral and good until the next call. This means that you must copy the returned bson if you wish to retain it beyond the lifetime of a single call to [mongoc_cursor_next()](https://mongoc.org/libmongoc/current/mongoc_cursor_next.html#).

`BindingsCursor::setSubstitution(const std::shared_ptr<Bindings> &bindings) (BindingsCursor.cpp:40)` passes a reference to the strings in the form of a `std::stringview` to the subclasses of `RDFNode`, which are then passed further to the pipeline.

To prevent the issue, the strings should be copied to a newly allocated place and freed when no longer in use.
I don't know the project structure enough to be able to know how this should be implemented. One possibility is to make subclasses of `Blank`, `IRIAtom`, and `StringView` that copy the string and free it when the objects are no longer referenced. However this only works, if the strings are not referenced elsewhere later for longer than the lifetime of these classes.

daniel86, 2024-05-15:
Thanks for spotting the bug. seems clear how this should be fixed:

- replace `StringView` by `StringCopy`
- use `IRIAtom::Tabled` instead of the constructor
- use `Blank::Tabled` instead of the constructor

PR #399 (2024-05-21): need to copy string data when iterating over mongo cursor
Closes #394
closed 2024-05-21 by daniel86