knowrob/knowrob #434: Dependency issue with hsrb packages installed

McModknower, 2024-09-03:
When using the toyota hsrb (which is used by the suturo project), you install some packages, including `libgazebo11-dev` which depends on `libcurl4-openssl-dev`. However `libraptor2-dev`, which is required for knowrob, depends on `libcurl4-gnutls-dev`.
These two packages conflict with each other (the both provide header files for libcurl-ssl), so you can't use the hsrb packages and knowrob together.

As a temporary fix i repackaged the libraptor2-dev package with a dependency on the more general `libcurl-ssl-dev`.
This works, but it is not clean.

Other potential solutions are:
- create an empty dummy package for `libcurl4-openssl-dev` to satisfy the dependency of `libpraptor2-dev` (optionally this dummy package depends on a real implementation)
- make the modified `libraptor2-dev` package available from a trusted source
  - contribute the change into the ubuntu package repository
- use the hsrb and knowrob on 2 different devices that are connected via (W)LAN
- remove raptor support from knowrob

daniel86, 2024-09-03:
thanks for the report!

Why did you skip the option to patch gazebo? I think this is more likely than patching raptor as I suppose the development is more active on the gazebo front.

McModknower, 2024-09-03:
> Why did you skip the option to patch gazebo? I think this is more likely than patching raptor as I suppose the development is more active on the gazebo front. Or wouldn't it be possible to patch the build files of gazebo?

I simply forgot to add that option. I added it now.

daniel86, 2024-09-03:
Another point: raptor and librdf are not part of the ROS package index.
Probably one of the reasons the dependency issue was unnoticed so far.

daniel86, 2024-09-03:
So I think best strategy would be to
- submit patches to gazebo and raptor and hope these will be included in Ubuntu packages at some point
- provide a repackaged version of raptor in the meantime, we could host it at the knowrob.org domain for Ubuntu 22.04 and 24.04

daniel86, 2024-09-03:
@McModknower would you be so kind sharing the repackaged version here?

hawkina, 2024-10-10:
are there any updates on this?

HoangGiang93, 2025-04-29:
Yes it's

HoangGiang93, 2025-04-29:
After spending some time troubleshooting with ChatGPT, I found a solution that allows installing the required dependencies **without uninstalling ROS, Gazebo, or any other existing packages**.
The key is to build `librdf` (Redland) and its dependencies from source manually.

Here’s the full process:

```bash
#!/bin/bash

set -e  # Exit immediately if a command fails
set -o pipefail

echo "🔍 Preparing to build Raptor2, Rasqal, and Redland from source."

# Install dependencies manually if missing:
# Uncomment the following line if you don't already have the required packages.
#
# sudo apt-get update
# sudo apt-get install -y \
#     build-essential     # Compiler tools: gcc, g++, make
#     autoconf automake libtool # Autotools for generating build scripts
#     flex bison           # For generating parsers (needed by Rasqal)
#     pkg-config           # Helper for managing compiler/linker flags
#     libxml2-dev          # XML parsing library (needed by Raptor2)
#     libpcre3-dev         # Regex support (needed by Rasqal)
#     uuid-dev             # UUID library (used by Redland)
#     libmysqlclient-dev   # MySQL client libraries (optional for MySQL backend)
#     wget                 # Tool to download source tarballs

# Create a working directory
mkdir -p ~/redland_build
cd ~/redland_build

# ==============================
# 1. Build and Install Raptor2
# ==============================
echo "📦 Building Raptor2..."
wget -c http://download.librdf.org/source/raptor2-2.0.15.tar.gz
tar xf raptor2-2.0.15.tar.gz
cd raptor2-2.0.15
./configure --with-openssl
make -j$(nproc)
sudo make install
cd ..

# ==============================
# 2. Build and Install Rasqal
# ==============================
echo "📦 Building Rasqal..."
wget -c http://download.librdf.org/source/rasqal-0.9.33.tar.gz
tar xf rasqal-0.9.33.tar.gz
cd rasqal-0.9.33
./configure --with-raptor2
make -j$(nproc)
sudo make install
cd ..

# ==============================
# 3. Build and Install Redland
# ==============================
echo "📦 Building Redland..."
wget -c http://download.librdf.org/source/redland-1.0.17.tar.gz
tar xf redland-1.0.17.tar.gz
cd redland-1.0.17

# Fix MySQL 'my_bool' compatibility issue
echo "🛠️  Fixing MySQL 'my_bool' issue..."
sed -i 's/\bmy_bool\b/bool/g' src/rdf_storage_mysql.c

# Configure, build, and install
./configure --with-raptor2 --with-rasqal
make -j$(nproc)
sudo make install

# Refresh linker cache
echo "🔄 Updating linker cache..."
sudo ldconfig

echo "✅ All components built and installed successfully!"

```