knowrob/knowrob #438: Build system incompatible with fmt 9+

mpomarlan, 2024-09-06:
When building QueryPipeline.cpp (and likely similar errors will show up later as well), an error will appear. The informative part is here:

/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt

libfmt version is 9.1 for myself. I suspect this will fail for version 10 as well, apparently there was a breaking change recently that

"removed deprecated implicit conversions for enums and conversions to primitive types for compatibility with std::format and to prevent potential ODR violations."

mpomarlan, 2024-09-06:
This actually seems related to an incompatibility between spdlog and fmt. Will investigate.

mpomarlan, 2024-09-06:
.

mpomarlan, 2024-09-06:
Reopened: not a version conflict between external packages. It is a KnowRob issue caused by knowrob::Predicate not being formattable for fmt 9.1.

There may be other classes affected, but this is where the build stops for me.

daniel86, 2024-09-06:
So I got the printing with fmt only running with code like this:
https://github.com/knowrob/knowrob/blob/dev/include/knowrob/formulas/Predicate.h#L81

Basically adding an instance of `<<` operator into the std::namespace. This seems hacky but was the only way I got it woring. clion also complains and warns about extending std namespace.

FYI fmt is used for expressions like `"foo {}", value` where the string form of value replaces the curly bracket at that position.

daniel86, 2024-09-06:
Any advise on how to do it correctly?

There are quite a few extensions of the std namespace for the `<<` operator in the code base. would be best if there is some way that also works with 22.04 fmt version (might be 8.32)

mpomarlan, 2024-09-06:
I think format_as needs to be overloaded for argument Predicate (or, Predicate const&). That would be the official easy fix, but it didn't work yet for me.

mpomarlan, 2024-09-06:
The harder fix is to define a formatter templated struct with Predicate as argument, and which contains a format method. Looks ugly and I don't think it would buy us anything, assuming format_as can be made to work.

daniel86, 2024-09-06:
Maybe it is after all some problem with interplay between spdlog and fmt.
The logging macros, e.g. `KB_INFO` are expanded here: https://github.com/knowrob/knowrob/blob/dev/include/knowrob/Logger.h#L32

It seems to me this `spdlog::default_logger_raw()->log` will do the fmt processing of the arguments. Maybe something is not right there. Maybe best to try first without these macros and spdlog

mpomarlan, 2024-09-06:
A similar problem is now thrown by QueryError(someMessage, aPredicate)

QueryError constructor does not use spdlog, just fmt

daniel86, 2024-09-07:
This PR might fix it:
https://github.com/knowrob/knowrob/pull/439

the fmt interaction is now handled by a base class: https://github.com/knowrob/knowrob/blob/fmt/include/knowrob/Printable.h#L55

this should make it easier in case it breaks again

closed 2024-09-07 by daniel86