knowrob/knowrob #447: Issue running queries with multiple results

Niamh-Ellis, 2024-10-11:
I'm trying to run an edited version of the intro tutorial in python. I have replaced the line: 
```
phi1 = QueryParser.parse("swrl_test:hasAncestor(swrl_test:'Lea', ?y)")
```
With 
```
phi1 = QueryParser.parse("swrl_test:isParentOf(swrl_test:'Rex', ?y)")
```
I have run this in the terminal getting both Ernest and Fred as results however running in python results in 
```
?y: swrl_test:Ernest
[13:56:29.395] [warning] Worker forced unwind
[13:56:31.400] [warning] Worker thread does not seem to exit, it will be detached!
```

daniel86, 2024-10-11:
Hi @Niamh-Ellis 

could you also provide the Python code?

In the tutorial example only the first solution is consumed. If there are more, the process will get interrupted once the result stream is deleted.
So to iterate over all results, you need to do something like this:
```
while True:
    nextResult = resultQueue->pop_front();
    if nextResult.indicatesEndOfEvaluation(): break
    ...
```

The interface looks a bit more complicated than before due to two reasons:
1. submitQuery returns before the evaluation is done. You get a kind of lazy queue where you need to pop results until it is indicated that no more results will follow. This also means you can process the first result while the second is still computed in some other threads.
2. the answer response can be *yes*, *no* or *don't know*, it can be marked as uncertain, and it can provide some support facts that can be used as an explanation. So it is more complicated then just the bindings for the query given.

hope that helps

Niamh-Ellis, 2024-10-16:
Hi @daniel86 
I understand now, thank you. 
What does the error I was getting actually mean?

daniel86, 2024-10-21:
The warning indicates that KnowRob has attempted to gracefully stop a worker thread but that this has failed. After some time that the thread does not respond to shutdown request, it will be detached instead.
It is very likely something related to the Python bindings, but it might not be critical as the shutdown only happens once you are done with processing the results

closed 2024-10-22 by Niamh-Ellis